ESPresense/ESPresense #457: Homeassistant autodiscovery on HA reboot

f0rdprefect, 2022-04-28:
When rebooting the HA instance (in my case the Raspi running HA in a docker container) all ESPresence devices are shown as unavailable after HA has started up again. 

By rebooting the individual devices they show up again. 

I guess the auto discover mqtt topic is only sent upon boot? 

Do you have a good way how to mitigate this?

DTTerastar, 2022-04-28:
I believe they are sent w/ retain which means home assistant will immediately get them.  I don't see this behavior at all, are you sure you haven't disabled retain or are expiring retain due to some configuration?

f0rdprefect, 2022-04-28:
Let me double check. 

I guess I know.

 The mqtt broker is running on the same raspi as the HA instance. So a reboot of the raspi clears of course the retained messages. Do you resent them in case the broker was unavailable?

DTTerastar, 2022-04-29:
I do not resend them and I don't think anybody else that uses the hass auto discovery does.  Can you make sure this is on: https://stackoverflow.com/questions/36126725/how-to-retain-messages-in-mqtt-if-the-mosquitto-broker-goes-down

f0rdprefect, 2022-05-01:
No I have not and after thinking about it I don't see how that would help
in case sensors are really missing.

Btw a similar issue was solved in the zigbee2mqtt project...

https://github.com/Koenkk/zigbee2mqtt/issues/105

Maybe it's not so strange after all?

DTTerastar, 2022-05-01:
I see talk of retain on birth/will and the discovery.  I think we currently use retain on all of those.  When you connect MQTT Explorer to your mqtt do you get the discovery immediately (I do!):

[image]

f0rdprefect, 2022-05-03:
I agree there is a lot of discussion in the zigbee2mqtt issue. 

What I meant was the comment which lead to [this commit](https://github.com/Koenkk/zigbee2mqtt/commit/cd13286ca7eda4a12bbca7a063e10636b6e3745e). Basically they send the autoconfig messages *again* on reconnect to the broker if it became unavailable before. 

If I was to save retain config messages, how would I get rid of those if I was to remove a device? Therefore I really like the zigbee2mqtt approach. In my point of view it really increases reliability. If I had more time I would have a look if I can produce a pull request. But that won't happen any time soon.

closed 2022-07-09 by stale[bot]