ESPresense/ESPresense #460: iPhone tracking not home

domemshur, 2022-04-29:
I’m tracking 2 iPhones and they seem to work for a little while, then they will say not home. The only way to change it is turning Bluetooth off and back on then it shows it’s in a room.

DTTerastar, 2022-04-30:
What kind of id are you using? (roomAssistant, apple:10*, apple:iphone*, exp:20)?

dwardu89, 2022-05-02:
I'm having the same problem, I'm using `apple:10*`

1liminal1, 2022-05-04:
Im also having the same issue. Even using the room assistant app, its either super slow or never changes from not_home 

Also if someone can answer me where apple:10* device type comes from/ how to recognise it. I tried matching up my phone BT mac to one of the devices that appear under apple:10* nothing though. So what is this?

dwardu89, 2022-05-05:
@DTTerastar is there anything i can do to give you more debugging information? it looks like the esp32 gets stuck at a query and just doesnt continue. i could share a copy of the terminal's output or i can set up a device with a more verbose installation?

needs-coffee, 2022-05-06:
hi @dwardu89 run with verbose logging and add the output here, what device are you running on (ESP platform), version, and what settings do you have configured in the device page?

DTTerastar, 2022-05-22:
If you use query and apple:10* you'll get a hit for only a second because the apple:10* is the id before you get a query response.  After the response you'll either get a roomAssistant or a apple:Modelnumber.  Makes sense?  We only send one id per mac, and the query "upgrades" the id from one type to another.

1liminal1, 2022-05-23:
Its not the model number LOL

joekeslar, 2022-06-14:
I'm using the Room-Assistant IOS app on my iPhone 12, filtering using the device ID from the room-assistant app, and am having the same issue where the phone isn't reporting properly after some period of time.

mrrobotnic, 2022-07-13:
Same thing here.  Two iPhone 13 Pros both have COVID exposure notifications turned on. One device will keep broadcasting without issue and the other will only broadcast when the screen is on.  Shortly after the screen turns off it stops broadcasting and becomes "not home".  I am using the apple:10*.  I also tried the Room Assistant app with the iPhone that stops broadcasting and ran into the same problem.

DTTerastar, 2022-07-18:
The newest beta has a feature to "Enroll" iOS devices. We store the BLE IRK and from that point on you can use that irk:{32chars} to uniquely identify that iOS device!

1liminal1, 2022-07-18:
That's awesome, Ive had to ditch BLE presence because I just cant seem to get my iPhone to keep an identifier that works :( 

When will we see this in the general release you think (roughly, a few weeks, a couple of months? so I can keep an eye out for it) 

Thanks

DTTerastar, 2022-07-18:
As long as the firmware is stable in my testing environment i'll release it in a few days.

gadric, 2022-07-18:
I am testing this, too. @DTTerastar Big Thanks!

1liminal1, 2022-07-19:
> I am testing this, too. @DTTerastar Big Thanks!

Any feed back yet?

DTTerastar, 2022-07-19:
The b1 allows you to just pick ESPresense in the Bluetooth pairing screen, no need to use rRF Connect.  Still can't get watches to encrypt and give up there IRKs.  iOS and Macs work great tho.

1liminal1, 2022-07-19:
> The b1 allows you to just pick ESPresense in the Bluetooth pairing screen, no need to use rRF Connect. Still can't get watches to encrypt and give up there IRKs. iOS and Macs work great tho.

woohoo! LOL

While the watch would be great, ill be happy with my phone :) 

What are detection speeds like? I was hopping it would be fast enough to transfer music to a different Alexa, maybe a big ask though.

DTTerastar, 2022-07-19:
We don't need to Query so the speed is the same as apple:10x fingerprints.  We just can tell them apart, finally!  IRKs are basically the "MAC" for random addresses.  Query still helps for Apple Watches, but otherwise it's useless now.

bananajoe86, 2022-07-20:
> The newest beta has a feature to "Enroll" iOS devices. We store the BLE IRK and from that point on you can use that irk:{32chars} to uniquely identify that iOS device!

Is there a guide or tip for integrating an iPhone? I have the latest beta, but I still can't get it right. I tried to register my iPhone Max in the IRK....
[image]

DTTerastar, 2022-07-20:
> Is there a guide or tip for integrating an iPhone? I have the latest beta, but I still can't get it right. I tried to register my iPhone Max in the IRK.... [image]

Mac addresses are useless when they are RPA (random private).  The IRK is a aes-256 key that you can use to figure out which random address is which.  The IRK is hard to find, I had to kinda hack into the esp32 ble controller to finally find it.  You can get it if you pair securely with that device.  So, in HAS you should see a button like this: 
[image] this will make your ESPresense discoverable via Bluetooth and then you can find and connect to it and once you do that you'll see irk:{32chars} as the id, etheir via serial or mqtt.  I'm storing the config for the devices as retained messages so that all nodes get the irks configured: 
[image]
which is kinda like hass autodiscovery, but for ESPresense itself.

DTTerastar, 2022-07-20:
Clear as mud?

bananajoe86, 2022-07-20:
Clear so far, thx. But how can I enroll without HassIO? Is there a mqtt topic to start this?

DTTerastar, 2022-07-20:
I'm going to change it so you can use a name instead of an id as the payload before release and it'll use the slugify as the id.

1liminal1, 2022-07-21:
That's so cool! Will the doco be updated at release to help us do that? Because I don't know what half of those words mean HAHA

joekeslar, 2022-07-22:
Just a tip for anyone trying to do this.  For me, the standard Apple bluetooth menu didn't show ESPresense as an access point after hitting the enroll button within HA.   I had to connect from my iPhone and iPad through a 3rd party tool like BLE Scanner or nFR Connect (both available from the app store).  Once connected, I used MQTT Explorer on my desktop to monitor the MQTT topics where I could easily get the IRK of each device.

I have 2 ESPresense devices and only enrolled to one of them.   I wasn't sure if I needed to do the enrollment process on both devices, or just one, but it seems that both ESPresense devices are publishing the info using the IRK value to MQTT. so it seems you only need to enroll to one of your ESPresense devices.  Please let me know if I'm wrong on that...

gadric, 2022-07-26:
I have this solution now running on 4 ESPs and tracking 1 iPhone 12 Pro and 1 iPad Pro for a couple of days. Works really good and very stable.

1liminal1, 2022-07-26:
So its released? I haven't got the update yet. Does it roll out by location?

DTTerastar, 2022-08-04:
The connection is meant to basically "steal" the irk.  Once we have the irk we can track the random macs passively.  Thus why it's so cool!  Yeah the above should work fine.

1liminal1, 2022-08-04:
Any news on release date?

DTTerastar, 2022-08-04:
I'm working on adding ui to the feature.  A button that says enroll that allows you to name the device.

1liminal1, 2022-08-04:
Ok thank you :)

1liminal1, 2022-08-05:
So I have 2.2.9, but no enroll button? Or its still coming? 

Thanks :)

DTTerastar, 2022-08-12:
Just allow prerelease and you'll have it.

Buckeyes1995, 2022-08-12:
Just found this thread and was able to enroll and find my iPhone 13 irk.  My HA is set up as follows. I'm having the same issue as the OP.  When I cycle my Bluetooth on the phone off/on the correct zone shows up in the state.  Walking across the house to the next ESP32, it doesn't change to that zone and times out to 'not home'.  When I walk back to the original ESP32, it never changes back without a Bluetooth cycle.

`- platform: mqtt_room
  device_id: "irk:7ea07914blahblahblah"
  name: 'Beacon - Jims Iphone'
  state_topic: 'espresense/rooms'
  timeout: 10
  away_timeout: 120`

joepv, 2022-08-18:
I have the same problem with an iPhone 13 (pro). If I restart the ESP32 in the room it becomes visible and after some time it goes offline. I use the IRK as id.

My Apple Watch is rock solid with the IRK.

1liminal1, 2022-08-18:
Are there any guides out there yet for the watch?

joepv, 2022-08-19:
> Are there any guides out there yet for the watch?

I'm working on it. Give me a few days.

What I found out. I removed the Espresense device from the Bluetooth settings on my iPhone after pairing. I think the iPhone removes/changes the IRK key at that point. _But it's a guess._ I now added the key from the Apple KeyChain (like the Apple Watch) and it looks more stable. But I have to test more over time.

1liminal1, 2022-08-19:
You're a star thank you :) 

Ive seen mentioned people are getting the IRK for the watch for Keychain, is that this app? 

[image]

Which section? 

Thanks again :)

joepv, 2022-08-19:
> You're a star thank you :) 
> 
> 
> 
> Ive seen mentioned people are getting the IRK for the watch for Keychain, is that this app? 
> 
> 
> 
> [image]
> 
> 
> 
> 
> 
> Which section? 
> 
> 
> 
> Thanks again :) 

Bluetooth, find the GUID, open the  settings. View password. Extract XML, get the key. Base64 decode it. Reverse it. Enter it in ESP.

Give me some time I'm documenting it with screenshots.

I think I was wrong about the iPhone. The beacon was stable until I got a phone call. After that the esp didn't find the phone until I restarted the esp. Very strange behavior. Another room picked up the phones so the beacon is broadcasted. It looks like there esp doesn't handle it well.

Buckeyes1995, 2022-08-19:
I've been unable to find my watch IRK.  I only have a windows machine, and for whatever reason, my Keys section under BTHPORT is completely empty.

DTTerastar, 2022-08-22:
> I've been unable to find my watch IRK. I only have a windows machine, and for whatever reason, my Keys section under BTHPORT is completely empty.

We need to figure out how to access the iCloud keychain without a mac.

Buckeyes1995, 2022-08-22:
Sorry, I’m really confused now.  I thought that article you pointed me showed me how to get it using RegEdit.. am I mistaken?  At the moment the only way is via a Mac?

If so, couldn’t I just run a MacOS VM to get it?

DTTerastar, 2022-08-23:
Yes, any way you can run MacOS and login via your apple creds should get you your iCloud keychain.  IRKs don't change unless you totally wipe your phone (it would unbind all your bluetooth stuff as well).  I have seen some wonkiness in the irk code during testing b7.  I'll look into it.

1liminal1, 2022-08-26:
> Clear as mud?

So I'm finally getting a chance to try this. So when you say "then you can find and connect to it" you mean in the bluetooth app on the phone, under settings right? 

It's not appearing for me, even though I can see it in the logs. 

ID: apple:1006:10-6                                           -39dBm

Im also on the iOS 16 public beta, if that makes a difference

AgentHawk, 2022-08-27:
Any news for a short Guide or tips how to get the IRK on a Mac?

joepv, 2022-08-27:
> Any news for a short Guide or tips how to get the IRK on a Mac?

Yes, I wrote a tutorial with screenshots, but the pull request is still not merged.

AgentHawk, 2022-08-27:
TY! It was easy to follow - but I cant find my Apple Watch via Bluetooth in the Keychain.
(Old but new installed Macbook Air)

joepv, 2022-08-27:
Strange, is the Macbook logged in with the same iCloud id? I found my watch id and my wife's after I created an account for her on my Macbook and she logged in with her iCloud.

AgentHawk, 2022-08-27:
yep - same ID

Keychain - iCloud - Bluetooth

only 2 "devices" shows up but without the "Public" and "random" stuff

joepv, 2022-08-27:
> yep - same ID
> 
> Keychain - iCloud - Bluetooth
> 
> only 2 "devices" shows up but without the "Public" and "random" stuff

I know old Macbooks do not support all continuity stuff like iPad screen sharing and with the AirPods. I think you need an M1 for that, but I'm not sure. With my old Macbook 2015 my AirPods behaved like standard bluetooth devices and with my newer M1 the continuity stuff works. I can also unlock my MacBook with my Apple Watch.

AgentHawk, 2022-08-27:
Uff^^

mrrobotnic, 2022-08-27:
How often does the check for updates run on the stable firmware (e.g. 2.2.9)? I want to play around with the latest version and set up the IRKs, but was not sure how long it would take after selecting the pre-release option to get it.  Looks like @DTTerastar just added in a manual update button a couple of days ago, which is awesome. Thanks for doing that!

mrrobotnic, 2022-08-28:
Was able to get the beta up and running and enroll two iPhones without an issue. Interestingly still seeing that one of the iPhones keeps broadcasting regardless of whether the screen is off or on and can always be seen by the various Espresense devices and the other iPhone stops broadcasting shortly after the screen turns off. Anyone else seeing similar issues?

1liminal1, 2022-08-31:
> So I'm finally getting a chance to try this. So when you say "then you can find and connect to it" you mean in the Bluetooth app on the phone, under settings right?

So my issue was, that I rebuilt the ESP from scratch to change networks and the Enroll button was still in HA, even though it did nothing. Deleted ESPesence from MQTT and the ESP from HA and now its working again with the iPhone IRK.... BUT, same issue as everyone else. If the iPhone leaves the area, disconnects. It comes back, but its quite slow (reboot is fast, seems like a couple of seconds to pick it up)

antongiacomo, 2022-08-31:
@mrrobotnic I have the same issue! The detection of an iPhone11 is quite reliable, sometimes a bit slow but pretty ok on average. The iPhone12 one is practically random. Both the detection are IRK based, no clue about the reason of this behaviour.

DTTerastar, 2022-09-01:
I found the issue.  Please try updating to the alpha via http://espresense.com/alpha

closed 2022-09-01 by DTTerastar

1liminal1, 2022-09-08:
So I know this is closed, but was that PR for the apple watch doco ever passed?

DTTerastar, 2022-09-08:
Yes, the documentation is live at https://espresense.com/beacons/apple

1liminal1, 2022-09-08:
Ahh, I was looking under configuration lol 

Thanks mate!