ESPresense/ESPresense #463: Huge Huge Delay

Axydas, 2022-05-02:
**Describe the bug**
I set up 4 pieces of ESP32 WROOM 32U as ESPresense stations and migrated them with Home Assistant. 

I am using two devices, one Garmin watch and one Android phone. 

Unfortunately,  the devices are unisuable due to a huge huge huge delay in detecting the devices. Also when it is detected the distance is not accurate. 

Some statistics; The average detection time for the Garmin is 105secs and for phone 120secs!!!!!!! 

Here I have to mention also that the WiFi signal of the ESP devices is OK if I flash them with any other firmware but with ESPresense firmware the devices are connected to WiFi only if I use them in the same room with my Mesh devices. 

I am wondering if I miss something or if I setup something wrongly. I need some help please. .

**Config**

- Version:
- Active scan enabled: Yes
- Include filter: No
- Exclude filter: No
- Query filter: No

**Hardware**

- Firmware flavor:
- Device (be specific): ESP32 WROOM 32U

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

needs-coffee, 2022-05-06:
hi @Axydas,
Is it only these 2 devices that you have long latency with? are there other devices you can test with?
can you have a look at the logs for the device and see when the device IDs are detected. 
I dont know about your specific models of devices but Bluetooth devices will advertise and respond at different intervals, if these devices are only broadcasting at 1.5 -2 minute intervals then Espresense can only update the location every 1.5 - 2 minutes.
did you [install a beacon app on your android device](https://espresense.com/beacons/apps)?

onlyoneme, 2022-05-06:
@Axydas 
You should inspect MQTT data received by the broker regarding intervals of broadcasted messages for your devices. Look for "interval" parts of each message.

oktay89, 2022-05-16:
I have exactly the same ESP and a Xiaomi Mi 10 Phone with Home Assistant app installed. The phone is 1m away from the ESP.
I debugged and here are the results: Interval on devices/iBeacon is between 9000 and 15000. RSSI is between -83 and -86. Distance is reported as 5.88m
I changed the ESP Calibration config:
RSSI Expected: -83
Absorption Factor: 3.5 (default)
Status LED - unchecked
All other settings on default.

Setting on the phone:
AdvertiseMode: lowPower
Measured Power: -59
Transmittter Power: low

What should I change to get updated more frequent and an accurate distance?

oktay89, 2022-05-16:
Ok, I have 3 options - low power(1hz), balanced(3hz), low latency(10hz).
Changing to "balanced" got me to intervals between 4200-6000. 
What's a good interval value I should be aiming for?
Also I noticed, when I close the Home Assistant App or do not move my phone for some time, no MQTT call is being sent and my presence changes to "not home". Maybe I should add some exception for the app in android?

onlyoneme, 2022-05-16:
@oktay89 
It's hard to say what's a good interval because it depends on your needs and the case you are implementing. If you want room level presence of frequently moving objects I would say 3 sec is a max.
Based on my experience there are some settings you can try to tune:
- sensor in HA:
  "away_timeout" - when the device is considered to be not_home if nothing is published on MQTT
  "timeout" - when to stop using old messages published on MQTT - lowering that value will make recognition faster but also more erratic
- ESP32 in ESPresense:
  "Factor for absorption" - increasing it will start considering tracked devices to be closer than they actually are
  "Skip reporting ..." - how frequent to publish state messages for stationary objects
  "Report early ..." - mesasured distance between 2 states to start considering the device as not stationary and publish states as 
  fast as possible

Most of those settings are closely related to advertise intervals of the tracked device - setting them below the device capabilities will not help at all.

oktay89, 2022-05-17:
Ok, changed RSSI expected from -59 to -83 and now the accuracy is a lot more realistic. So issue 1 is solved, but the device still stops sending messages after few minutes and HA shows me as "not_home". 
away_timeout = 120, 
timeout = 5
Skip reporting tried default and 2000
Report early tried default and 0.1

onlyoneme, 2022-05-17:
@oktay89 
I have no experience with HA companion app and beacon simulation within it. You can try any other available beacon app. 
Maybe beacon service is being killed by the phone when device is idle.

closed 2022-07-23 by stale[bot]