ESPresense/ESPresense #464: Switch to a different WiFi network?

uspino2, 2022-05-03:
This has been asked before but I can't find a clear answer. Is there a way to move the ESPresense module to a different WiFi network, other than re-flash and start from scratch at the captive web portal?

Thanks.

dwardu89, 2022-05-05:
if you need to do that, probably the best is to block that device from the wifi network, or get it out of range  of that wifi AP. it will create it's own wifi network, and connect to that and change the settings like that

DTTerastar, 2022-05-23:
Yeah, either kill the wifi network you don't want it to use, or erase and reflash.  On certain variants it does support holding down a button to erase (ala a Wifi router).

closed 2022-07-29 by stale[bot]