ESPresense/ESPresense #465: M5stack ENV III sensor

RienduPre, 2022-05-04:
I am using 8 M5stacks as esprecense devices and like to add the M5Stack ENV III sensor for temperature and humidity reporting. 
is this sensor supported?

thanks
Rien

RienduPre, 2022-05-08:
Some extra info which may help:

The M5Stack ENV III has a SHT30 and QMP6988 sensor to detect temperature, humidity, and atmospheric pressure data.  They support the I2C interface (SHT30:0x44 , QMP6988:0x70).

DTTerastar, 2022-06-04:
I'd need one myself to try and do it.  I have so many other ways of telling temp and humidity it's fairly low on my list of stuff to do.

DTTerastar, 2022-08-13:
I have a ENVII and it had a SHT30 on it, so I was able to add half of this.

krazos, 2022-09-12:
I am pleased to report that temperature and humidity readings from my M5Stack ENV III sensor are working with ESPresense using the SHT30 configuration and I2C address 0x44. I am not overly concerned with the lack of air pressure readings, but I may eventually experiment and see if I can get the QMP6988 up and running.

RienduPre, 2022-09-13:
Great!!
Tested it and it works :-)

Many thanks for this update.

krazos, 2022-11-12:
Keeping this issue alive pending support for the QMP6988 atmospheric pressure sensor.

radkokeves, 2023-02-10:
If anyone is interested in implementation of OOTB example, here it is: https://github.com/radkokeves/ESPresense_m5stickcplus_sgp30-baseline/src/m5envIII.cpp

dxmnkd316, 2024-07-16:
I attempted the same thing with the ENV IV sensor.  I wasn't able to get the SHT40 sensor working at first, but made a few changes in the firmware and it worked.  Will do some more testing to see if this is because I chose a different build (CDC) or if it was the code changes.

However, I was unable to get the BMP280 to work.  The troubleshooting i've done seems to show that the Adafruit firmware isn't initializing the BMP with the begin command.  Not entirely sure what's going on there.