ESPresense/ESPresense #467: vindriktning particle sensor

cranky-coder, 2022-05-06:
Has anyone attempted to add the vindriktning particle sensor to espresense?  Ikea has a small airquality sensor box that has this sensor that MANY have already hacked and have the information being presented to home-assistant and the like.  However, I like the idea of adding this sensor to espresense.  I have some modules built already that have temp/humidity and light levels.  Adding in air quality and motion (which i see is already supported) would make these perfect multi-sensors for every room.

I have been looking through the code a bit, are there any further developer docs other than the building.md file?  Like how to add things to the UI?  

Either way I am going to give it a shot.  Just wondering if maybe someone has already started down this route.

DTTerastar, 2022-05-06:
This is the docs for the UI: https://github.com/Juerd/ESP-WiFiSettings

I don't know of anyone adding this, sounds cool!

cranky-coder, 2022-05-06:
Oh sweet!!! Thanks for that.   Ill start playing with it a bit this weekend and see what I can do.  If I can bumble it together Ill do a PR and have you guys take a peek!  Thanks!

closed 2022-07-12 by stale[bot]