ESPresense/ESPresense #468: Auto Update over-written 2.1.1 with 2.1.0.

JohnTranQUT, 2022-05-09:
**Describe the bug**

2.1.0 update has over-written 2.1.1 update.

**Config**

- Version: 2.1.1 ESPresense with Sensors converted over to 2.1.0 automatically

**Hardware**

- Firmware flavor: ESP32 with Sensors
- Device (be specific): ESP32

**Logs**

No logs however I've noticed my HX711 weight sensors weren't working. After rechecking my wiring, nothing was notably wrong. Checked Home-assistant logs, and noted that about 10:00AM UTC 9th May.

I've noticed my ESP32 device was on version 2.1.0 and the webpage shown no hx711 sensor settings.
Recalled the version involving hx711 being an increment higher. Github shows that the latest version is 2.1.0 now. It was 2.1.1 about 26 days ago.

Also, there is 2.1.1 here
https://github.com/ESPresense/ESPresense/tags
however, it does not show up when going to
https://github.com/ESPresense/ESPresense/releases

**Screenshots**

[image]

onlyoneme, 2022-05-11:
2.1.1 is back. I think that the switch to disable/enable possible downgrades during auto ota would be nice.

JohnTranQUT, 2022-05-13:
> 2.1.1 is back. I think that the switch to disable/enable possible downgrades during auto ota would be nice.

Looks like you are right. All of my ESP32 Presence had been updated overnight.
This issue can be closed now.

closed 2022-05-13 by JohnTranQUT