ESPresense/ESPresense #469: Reboot Loop ESP-WROOM-32

burnsie-la, 2022-05-10:
Overnight two of my 5 ESPresense devices got stuck into a reboot loop (They were both working fine when I set them up yesterday).  They both faced this error:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (78:E3:6D:10:B5:A8)...
Connecting to WiFi SSID 'burnsie_iot'.....................................172.16.1.104
Version:    v2.0.37
IP address:   172.16.1.104
DNS address:  192.168.1.3
Hostname:     espresense-living-room
Room:         Living Room
MQTT server:  192.168.1.3:1883
Max Distance: 5.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
1 New   | MAC: 68d79a26b374, ID: sd:0x252a                                                    
1 New   | MAC: f23d0a25d8fc, ID: f23d0a25d8fc                                                 
1 New   | MAC: 63232348f987, ID: sd:0xfe9f                                                    
abort() was called at PC 0x401aab1f on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f670:0x3ffe5d10 0x4008f8ed:0x3ffe5d30 0x401aab1f:0x3ffe5d50 0x401aab66:0x3ffe5d70 0x401aa23f:0x3ffe5d90 0x401aa576:0x3ffe5db0 0x401af771:0x3ffe5dd0 0x401af8e6:0x3ffe5df0 0x401afa11:0x3ffe5e30 0x400f7dcd:0x3ffe5e50 0x400fa56f:0x3ffe5e70 0x400f60a2:0x3ffe5eb0 0x400f6287:0x3ffe5f40 0x400d2492:0x3ffe5f60 0x400e1121:0x3ffe5f80 0x400e45da:0x3ffe5fc0 0x400e94b5:0x3ffe6030 0x400e9561:0x3ffe6070 0x400e97ad:0x3ffe6090 0x400e8065:0x3ffe60b0 0x400ef1d6:0x3ffe60d0 0x400e065b:0x3ffe60f0 0x400908fe:0x3ffe6110

Rebooting...`

iCSpotRun, 2022-05-10:
You seem to be running an old version (v2.0.37). v2.1.1 has a few fixes for reboot looping.

burnsie-la, 2022-05-10:
Oh cool - how do I update to a newer version? I went to the ESPresense website yesterday and used the tool on the firmware page to install it.

JohnTranQUT, 2022-05-10:
> Oh cool - how do I update to a newer version? I went to the ESPresense website yesterday and used the tool on the firmware page to install it.

As of right now, the website and GitHub releases appear to have been defaulted to the 2.1.0 version because of the issue https://github.com/ESPresense/ESPresense/issues/468.

**Like me, if you are after 2.1.1 specifically...**

You would need to use the alternative method of ESP update, where you download the bin files from the releases below link and then manually update via the ESP firmware updater program. This option would be covered in the documentation and you will lose all previous configuration settings:
https://github.com/ESPresense/ESPresense/tags

You would also need to disable auto-update for now (to ensure 2.1.0 downgrade doesn't happen) or enabled auto-update with allowance for pre-releases (it will downgrade to no lower than 2.1.16b - I think, that should have the looping problem solved).

onlyoneme, 2022-05-11:
@burnsie-la 
I've found some entries in MQTT related to fw 2.0.37 and generated yesterday. Also "ota_update" switch has appeared in MQTT discovery which has been renamed to "arduino_ota" in later fw. I guess your issues are the result of auto downgrade incident mentioned in #468.

closed 2022-05-11 by burnsie-la