ESPresense/ESPresense #471: WiFi reconnect takes more than 10 minutes.

gadric, 2022-05-15:
**Describe the bug**

In my current setup, my wifi will be switched off, when everybody leaves home. And it will be turn on again, if someone comes home again. I am currently seeing that it takes more than 10 minutes before the ESPs are rejoining the network. I thought it should be faster. 

**Config**

- Version: 2.1.1
- Active scan enabled: yes
- Include filter: apple:1006:10-8 
- Exclude filter: no
- Query filter: apple:1006:10-8

**Hardware**

- Firmware flavor: esp32.bin
- Device (be specific): AZDelivery ESP32 Dev Kit C V4 NodeMCU

DTTerastar, 2022-06-04:
The code pops the wifi portal if it can't connect but I believe that only lasts 1 minute.  I use an espresense in my car and it quickly finds the wifi when I drive in, so i'm not sure what's different about your setup?

DTTerastar, 2022-07-18:
I do pop the portal for much longer.  I can make that configurable.

closed 2022-09-23 by stale[bot]

errolsancaktar, 2023-06-24:
> The code pops the wifi portal if it can't connect but I believe that only lasts 1 minute.  I use an espresense in my car and it quickly finds the wifi when I drive in, so i'm not sure what's different about your setup?

Sorry that this is off topic but can you explain how you use it in your car?