ESPresense/ESPresense #474: Duplicate devices ID - Apple Watches

clipse2004, 2022-05-17:
**Describe the bug**

Hello,
I have the problem that I can track my two different Apple Watches.
If I query via the terminal and go once with the Watch 4 and then Watch 6, I have the same ID for both. However, one is from my girlfriend and one from me ^^

Apple Watches:
 device_id: "apple:1005:9-24"

iPhones:
  device_id: "apple:1007:11-8"
  device_id: "apple:1006:10-6"

The iPhones are displayed differently, for example, that fits.

Is that normal? 
Can not somehow track via MAC? These are namely different.

**Config**

- Version: 2.1.1
- Active scan enabled: No
- Include filter: No
- Exclude filter: No
- Query filter: No

**Hardware**

- Firmware flavor:
- Device (be specific): 
   ESP32 Dev Kit C V4 NodeMCU

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

clipse2004, 2022-05-18:
okay, I do the query on the Esp32 interface, right? 
Do I then specify the more accurate ID in Homeassistant with the sensors instead of the old Id?
Or is that then defined on the esp devices?

clipse2004, 2022-05-18:
Thank you for the explanation, I understood that now :) 

Gibts still an alternative if the query fails? 
With the iPhone it goes but unfortunately with the Watch always an error :/

onlyoneme, 2022-05-18:
If querying for specific service data is not possible all you can use is counting, just to get the number of unique devices hidden behind the same id. It has been explained here: 
https://espresense.com/configuration/settings

clipse2004, 2022-05-18:
So a query fails on both watches, always query timeout. 
Now I have tried to put both on one sensor, that works but if I am in the living room and the lady in the kitchen, then it always jumps back and forth. 

The counting I have also tried, but if I understood correctly, then he counts the devices and summarizes them together right? 
I would prefer if all devices have their own sensor. 

I must now ask again, why is a filter via MAC not possible? They are basically fixed and different, would not that be the easiest? 

I already thought about doing this without the watches.  Problem here, you have the watch more often on the wrist than the phone in the pocket :D

clipse2004, 2022-05-18:
Ah okay in the MQTT Explorer I could now read out data to the clocks. 
Different are e.g. the two "disk" parameters, can I use these instead of the ID?

so as an example device_id: "2518fe95a8"?

DTTerastar, 2022-06-26:
> So a query fails on both watches, always query timeout. Now I have tried to put both on one sensor, that works but if I am in the living room and the lady in the kitchen, then it always jumps back and forth.

The latest beta fixes query for watches.  Please test it!