ESPresense/ESPresense #475: Radioland iBeacon shown in log, but not in mqtt

JojoS62, 2022-05-19:
I have a nRF51822 based button from Radioland, and it advertises every second as an iBeacon. Via app, I can set also the name of beacon:
```
Verbose | name:jojo                                                 AD: 00001803-494c-4f47-4943-544543480000
Verbose | name:jojo                                                 MD: 4c000215fda50693a4e24fb1afcfc6eb076478250064ffff1e
1 New   | MAC: df51fd4a4caf, ID: iBeacon:fda50693-a4e2-4fb1-afcf-c6eb07647825-100-65535  
```
But this device is not added to the mqtt devices. What can I do get it listet?

JojoS62, 2022-05-19:
now I have compiled the sources and found when I omit
https://github.com/ESPresense/ESPresense/blob/4e0a39d77cf76595555fcb249987b20c5b4dab08/lib/BleFingerprint/BleFingerprint.cpp#L291
then the iBeacon is added to the list. The beacon advertises a power value of 32 (result of getSignalPower()), maybe this is read wrong?

DTTerastar, 2022-05-23:
You're supposed to "calibrate" iBeacons by reading the rssi@1m and reprogramming it with the correct value.  ESPresense does read and honor the value that is sent.  Is there an app it came with to reconfigure it?

JojoS62, 2022-05-23:
Thanks, yes, I have an app to change some beacon values. But the sent RSSI is depending on the battery level as I understand from the documentation. I have added a printf for the getSignalPower and its constantly sending a value of about 30. When this is written to calRSSI, then the beacon is not added.

DTTerastar, 2022-06-04:
We could make an "override iBeacon rss@1m" option I guess

DTTerastar, 2022-06-24:
Is this it? https://fcc.report/FCC-ID/2AGUT-NRF52810/4587802.pdf

JojoS62, 2022-06-25:
Yes, exactly

DTTerastar, 2022-06-26:
[image]
 This is the setting to change.

closed 2022-09-01 by DTTerastar