ESPresense/ESPresense #477: Support sonoff m5

RaZer0r, 2022-05-20:
**Is your feature request related to a problem? Please describe.**
I’m about to install my dual button m5 and would love to run espsense as well to further hide anything tech in the house ;). 

**Describe the solution you'd like**
support for buttons like on the sonoff m5

**Describe alternatives you've considered**
I could run esphome, but the BLE integration just isn’t as good.

DTTerastar, 2022-05-23:
So we need to support relays and buttons?

RaZer0r, 2022-05-23:
yes, relays, buttons and leds (I'm not sure if the on/off led is controllable separately on each switch/button)

RaZer0r, 2022-06-07:
Where are you located? If possible, I would definitely consider sending you a unit

DTTerastar, 2022-06-18:
Pennsylvania, USA

Mdleal, 2022-07-15:
I really want relay support for general relay use, I could send you money or ship you one from amazon to support the cause.  LMK.

Mdleal, 2022-07-18:
Can you add the M5 and not the nspanel?

DTTerastar, 2022-07-18:
Didn't even know there were the non screen ones :)  Done.

Mdleal, 2022-07-18:
Ordered the 3 gang which is only 1 gang with 3 switches which is weird they call it 3 gang. Should be there tomorrow.

DTTerastar, 2022-07-18:
Cool

bqtran, 2022-08-30:
> https://www.amazon.com/hz/wishlist/ls/MTFZJ03S78XY?ref_=wl_share I believe you can just ship it to Your Name’s Gift Registry Address via this.

Do you still need the hardware to get this going? I also don't mind footing the bill for a dev unit or two. :)

DTTerastar, 2022-08-30:
I have it.  The 3 led support in the new v3 is step 1 of supporting it...

RaZer0r, 2022-09-02:
Nice ;)

DTTerastar, 2022-09-15:
Sorry, this keeps getting back burnered.

closed 2022-11-21 by stale[bot]

RaZer0r, 2022-12-06:
Can this be kept on the schedule?