ESPresense/ESPresense #480: Too frequent MQTT msgs

Vendo232, 2022-05-23:
**Describe the bug**

Question: Started using the ESPresense on ESP32 for room presence detection and noticed in MQTT Explorer the MQTT msgs are pinged or sent multiple per second , could this frequent MQTT traffic influence WIFI throughput , create wifi noise and mainly is it really necessary to send msqs ever second?  Is it possible to reduce the MQTT msg frequency?

**Config**

- Version: 2.1.1
- Active scan enabled:
- Include filter:
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: ESP32 mini
- Device (be specific):

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**
[image]

[image]

onlyoneme, 2022-05-24:
You can focus on selected devices using Include / Exclude filtering. You can control the frequency of MQTT messages with Report early / Skip reporting settings.

DTTerastar, 2022-06-04:
mqtt_room needs constant messages, it's just how it works.  Once we do more processing in ESP this could be made better, but until then not sure how to make it better other than the features mentioned by @onlyoneme above.

bogorad, 2022-07-15:
I'm also monitoring BLE Xiaomi temp/humidity sensors, and although the sensors report only once in 10 min, MQTT is constantly spammed with this data:

https://imgur.com/a/jJ1UID3

onlyoneme, 2022-07-15:
@bogorad 
Temperature or humidity monitoring are special cases which are not a part of main ESPresense purpose - movement and presence tracking, which requires constant messaging. If the frequency of messages published to MQTT is  annoying for you, I would suggest to define a dedicated esp32 device which would track only those mentioned sensors. This would allow you to limit the frequency on the host level.

DTTerastar, 2022-08-04:
You can reduce the messages by half by using either devices or rooms not both.  Try that and see if it helps.

closed 2022-10-11 by stale[bot]