ESPresense/ESPresense #482: HA not updating

droidsteve, 2022-05-25:
Home Assistant showing device "Not_home" and "distance: null"

**Config**

- Version:
Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
espresence - current version
- Active scan enabled: yes
- Include filter: n/a
- Exclude filter: n/a
- Query filter: n/a

**Hardware**
- Device (be specific): ESP32

mqtt explorer:
"id": "iBeacon:00b67fe9-b8e5-491e-b697-8fc0bf2c3eca-0-0",
  "idType": 175,
  "rssi@1m": -65,
  "rssi": -67,
  "raw": 1.14,

HA Dev Tools:
sensor.s20_bleS20 BLE         | not_home             | distance: null

Sesnor.yaml
  - platform: mqtt_room
    name: 'S20 BLE'
    device_id: "iBeacon:00b67fe9-b8e5-491e-b697-8fc0bf2c3eca-0-0"
    state_topic: "espresence/rooms"
    timeout: 5
    away_timeout: 120

DTTerastar, 2022-06-04:
It's espresense/rooms.  You spelled presence not PRE SENSE.

simenopsahl, 2022-07-10:
LMAO, I had the excact same problem. If not for this I would probably never have figured it out 👍

hellcry37, 2022-07-13:
I think this can be safely closed as resolved