ESPresense/ESPresense #490: Distance and Latency settings for mmWave Radar

EvisHome, 2022-06-09:
**Is your feature request related to a problem? Please describe.**
An issue I have when using ESPresense with mmWave Radar sensor is that these sensor detects movement through walls, or bounce the signal to next room, so I cannot get a room presence as such, to be able to use it in room automations.

**Describe the solution you'd like**
Would be nice if in the settings you could change the Distance setting, also "cool-off" latency adjustment would be nice.

**Describe alternatives you've considered**
I am currenty using ESPhome for the mmWave radar, to have the distance and latency settings, but it's not optimal to have another ESP32 in the room for just the Radar, while I could all be in the same solution .. would be cool to have PIR for fast initial motion detection, the Radar for presence detection and the BLE tracking for "who" .. I am tracking only BLE wearables (Oura Rings gen3) and that works really nicely in ESPresense for certain devices.

**Additional context**
The solutions I am using currently for mmWave Radar
https://community.home-assistant.io/t/mmwave-presence-detection-esphome-style/382778

[image]

I have designed a 3D printable cases that can fit in ESP32 DEVKIT DOIT with PIR (AM312) and mmWave Radar (SEN0395) (free to use)
[image]

[image]

[image]

and a plain case for just ESP32 DEVKIT DOIT board
[image]

DTTerastar, 2022-06-26:
Out of stock everywhere...

webhostsg, 2022-07-04:
> Out of stock everywhere...

Maybe try this alternative? --> https://www.aliexpress.com/item/1005004139309359.html?

DTTerastar, 2022-07-04:
https://www.aliexpress.com/item/3256803953177606.html has free shipping. Is that the same one?

webhostsg, 2022-07-04:
> https://www.aliexpress.com/item/3256803953177606.html has free shipping. Is that the same one?

Both are the same, prices are different. Just one is free shipping (slightly more expensive) and the other is not (slightly cheaper). Depends on where you ordering from the Free shipping version could be cheaper.

webhostsg, 2022-07-04:
> Both are the same, prices are different. Just one is free shipping (slightly more expensive) and the other is not (slightly cheaper). Depends on where you ordering from the Free shipping version [...]

Confirm the same by the seller.
[image]

webhostsg, 2022-07-18:
> Ok, 6 weeks and i'll let you know...

Another one you might consider integrating. A radar sensor with fall, heartbeat and breathing detection capability.

https://wiki.seeedstudio.com/Radar_MR60FDA1/

DTTerastar, 2022-07-19:
Why are there separate modules, one for fall and one for heartbeat?  And totally different prices?

webhostsg, 2022-07-19:
> Why are there separate modules, one for fall and one for heartbeat? And totally different prices?

For the 60GHz only two versions, they are the upgraded versions of the 24GHz versions. I think the Fall Detection version would be more popular
[image]

summongod, 2022-07-22:
I just hacked in the serial only sensor mentioned above. Seems to work ok. Can share the code but it is fugly

summongod, 2022-07-22:
I also have the manual from the vendor which I can post if anybody is interested.

summongod, 2022-07-22:
[HLK-LD1125H-24GApplication Manual.pdf](https://github.com/ESPresense/ESPresense/files/9169056/HLK-LD1125H-24GApplication.Manual.pdf)

webhostsg, 2022-07-26:
Another affordable and simple mmwave from HLK that could have its sensing distance and sensitivity adjusted.
HLK-LD2410
https://www.aliexpress.com/item/1005004351593073.html
[HLK-LD2410 user manual V1.02.pdf](https://github.com/ESPresense/ESPresense/files/9189632/HLK-LD2410.user.manual.V1.02.pdf)

summongod, 2022-07-26:
It would be interesting to see how accurate the detection is when compared with the module I have which outputs serial data that can be used to implement logic in ESPresense. The manual doesn't seem to document the serial interface commands either which is a shame. Cost is good and I noted thant HLK seem to be adding new cheap boards quite frequently.
I wonder if HLK driving down the cost of these 24G radars will mean the end for PIRs especially for room presence detection?

webhostsg, 2022-07-26:
> It would be interesting to see how accurate the detection is when compared with the module I have which outputs serial data that can be used to implement logic in ESPresense. The manual doesn't seem [...]

Yup, I did email them about the configuration commands thru serial, still waiting for a reply.

webhostsg, 2022-08-01:
7 left in stock. Hurry!
[image]

DTTerastar, 2022-08-04:
I have one now too... need to play... gotta finish irk stuff first tho...

webhostsg, 2022-08-12:
> It would be interesting to see how accurate the detection is when compared with the module I have which outputs serial data that can be used to implement logic in ESPresense. The manual doesn't seem [...]

The serial interface commands:
[HLK-LD2410 Serial Communication Protocol V1.02.pdf](https://github.com/ESPresense/ESPresense/files/9312938/HLK-LD2410.Serial.Communication.Protocol.V1.02.pdf)
 You could actually configure the device and after that just use it with a single GPIO using the OUT pin. Currently using it with the default config, seems ok for my needs. The only thing I dislike is it comes pre-soldered with header pins and can't seem to find the right connector for 1.27mm pitch. Soldering wires directly to the pins is a pain.
My Setup, fits nicely on a 86x86 blank faceplate:
[image]
[image]

crlogic, 2022-08-12:
@webhostsg Thanks for the serial command pdf. HLK don't make it easy to find them.

Just grabbed two 2410's and am dreading the soldering.  At least with distance and sensitivity it looks like a potentially close match for the DFRobot.  Parsing the data is going to interesting..

webhostsg, 2022-08-12:
> @webhostsg Thanks for the serial command pdf. HLK don't make it easy to find them.
> 
> Just grabbed two 2410's and am dreading the soldering. At least with distance and sensitivity it looks like a potentially close match for the DFRobot. Parsing the data is going to interesting..

They don't disclose it when you email them. They will message on AliExpress asking for your email address when you purchase. I missed their message on AliExpress initially. I'm trying to find a connector for the tiny 1.27mm pitch pins, too close to solder, even 0805 SMDs are not that hard to hand solder.

webhostsg, 2022-08-12:
The configuration tool for LD2410 just in case someone needs it:
[LD2410 Tool EN（英文版）.zip](https://github.com/ESPresense/ESPresense/files/9315957/LD2410.Tool.EN.zip)

webhostsg, 2022-08-16:
@crlogic sourcing a 1.27mm pitch connector is not easy as well. I guess I'll just create a small PCB to convert it to normal pin headers.
[image]

crlogic, 2022-08-16:
@webhostsg - that looks excellent! I think I killed one trying to remove the headers and solder tiny wires with my only (fat) tip..   ..the second try by soldering directly to the headers worked but isn't pretty.

webhostsg, 2022-08-16:
@crlogic I ordered 30 pcs of the PCB as it only cost 50 cents more than ordering 5 pieces. I just maybe need 6-10, I'll send you a few if you need them, maybe just pay for postage.

crlogic, 2022-08-16:
@webhostsg deal. crlogic [at] ymail [dot] com

webhostsg, 2022-08-16:
@crlogic will email you once they are in.

summongod, 2022-08-16:
I'd also be happy to buy a few from you and also pay for the shipping. This definitely less risky than soldering with my basics skills.

webhostsg, 2022-08-16:
@summongod ok will update you once they have arrived. You would still need to solder the pins from the LD2410 to the PCB I created and the header pins as well. Just much easier to solder them instead of wire direct to the LD2410 pins.

[image]
[image]

crlogic, 2022-08-17:
@webhostsg that Protocol PDF is a mess..

https://imgur.com/a/oXV5cXn

crlogic, 2022-08-17:
> HLK that could have its sensing distance and sensitivity adjusted.

With distance sorted, the next step is adding sensitivity adjustment. Yet there is a crux. In the protocol pdf, there is a sensitivity chart and it changed per distance "step".

So while it would be simplest to configure a global sensitivity setting. Why have HLK gone with a decrementing approach?  Too many false positives otherwise?

I think I'll start with global and see where that lands.

webhostsg, 2022-08-17:
@crlogic was wondering that too... diminishing sensitivity the higher the gate. But i have one in the kitchen with default config, seems to work well.
[image]
[image]

crlogic, 2022-08-17:
> I need to adjust the sensitivity for gate0 or gate1 for one of the room since the wall fan triggers motion. Need to wait for the PCB before adjusting it.

I haven't tried individual gate adjustments yet.

I installed the LD2410 beside a DFRobot to monitor sleeping/bed occupancy last night and it wasn't quite right.  That is stock sensitivity though.

I added "all gates" sensitivity this morning and it works as advertised.  I have figured out that the sensitivity value is basically the "energy threshold" for detection.  So you can use the energy values to figure out how to adjust.

[image]

I can't quite figure out the difference between the distances; especially detectDistance vs Still/Movement.  They are not accurate by any means and more a "rough" measurement.

webhostsg, 2022-08-18:
@crlogic great progress, hope this could be implemented in ESPresence.

PCB progress:
[image]

DTTerastar, 2022-08-18:
The DFRobot sensor is finally back in stock so I ordered 2.  I still need to wire up and test the cheaper one I did get as well.

webhostsg, 2022-08-18:
`In addition, if the sensitivity of a certain distance gate is set to 100, the effect of not recognizing the target under the distance gate can be achieved. For example, suppose the sensitivity of distance gate 3 and distance gate 4 is set to 20, and the sensitivity of other distance gates is set to 100. In that case, it is possible to detect only the human body within the range of 2.25-3.75m from the distance module.`

So if gate 3 and gate 4 are set to 20, all other gates are automatically set to 100 and human presence/motion is only detected within gate 3 and gate 4 (2.25-3.75m). The lower the number (for sensitivity aka energy) the higher the sensitivity. No wonder the default config chart shows diminishing values the further the distance.
[image]

crlogic, 2022-08-18:
> Update on PCB progress:

Wow that's quick!

crlogic, 2022-08-19:
> Would also mean that we would need to configure all gates, else it would set only the gates that were configured and change all other gates to 100 (essentially disabling other gates)

Currently this is the approach (setting all values equal). Unclear how to easily present so many different gates for configuration.  I am also working off ESPHome. Hopefully that code can work in ESPresense.

I also wouldn't read too much into the protocol guide. It has some serious translation issues and wrong in numerous packet examples (wrong byte Len etc..)

webhostsg, 2022-08-19:
> Currently this is the approach (setting all values equal). Unclear how to easily present so many different gates for configuration. I am also working off ESPHome. Hopefully that code can work in [...]

Yes, I think if it would be implemented into ESPresence just the simpler configuration would be enough. Like distance and overall sensitivity (all gates equal). If someone wanted to configure each individual gate they could hook up directly to the serial of the radar radar sensor and config it and plug it back into ESP32

webhostsg, 2022-08-22:
PCB update (arrived at my country's airport):
[image]

qkddn49, 2022-08-22:
Can I get the espome code for LD2410

crlogic, 2022-08-22:
> Can I get the espome code for LD2410

https://github.com/rain931215/ESPHome-LD2410

webhostsg, 2022-08-23:
> > Can I get the espome code for LD2410
> 
> https://github.com/rain931215/ESPHome-LD2410

@crlogic I have not tried your code, but looking at it I'm a bit confused about `Max Moving Distance Range` and `Max Still Distance Range`. 

```
  - platform: template
    name: "Max Moving Distance Range"
    id: maxMovingDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(x,id(maxStillDistanceRange).state,id(noneDuration).state);
  - platform: template
    name: "Max Still Distance Range"
    id: maxStillDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,x,id(noneDuration).state);
 ```
 why is the `...id(maxMovingDistanceRange)...`  for `name: "Max Still Distance Range"` and not `x,id(maxStillDistanceRange`

crlogic, 2022-08-23:
The x variable is used in the lambda for the number being changed. But the sensor needs other data from the other numbers as well.

webhostsg, 2022-08-23:
> The x variable is used in the lambda for the number being changed. But the module (protocol) needs other data from the other numbers as well.

@crlogic thank you. I just figured it out after looking through the code again. Missed it earlier.

webhostsg, 2022-08-23:
@crlogic for individual gate sensitivity

ld2410.yaml:
```
number:
  - platform: template
    name: "Set Gate"
    id: setDistanceGate
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setGateSensitivity(x,id(setMotionSensitivity).state,id(setStillSensitivity).state);
  - platform: template
    name: "Set Motion Sensitivity"
    id: setMotionSensitivity
    min_value: 0
    max_value: 100
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setGateSensitivity(id(setDistanceGate).state,x,id(setStillSensitivity).state);      
  - platform: template
    name: "Set Still Sensitivity"
    id: setStillSensitivity
    min_value: 0
    max_value: 100
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setGateSensitivity(id(setDistanceGate).state,id(setMotionSensitivity).state,x);
```

ld2410_uart.h:
```
  void setGateSensitivity(int setDistanceGate, int setMotionSensitivity, int setStillSensitivity)
  {
    char cmd[2] = {0x64, 0x00};
    char value[18] = {0x00, 0x00, lowByte(setDistanceGate), highByte(setDistanceGate), 0x00, 0x00, 0x01, 0x00, lowByte(setMotionSensitivity), highByte(setMotionSensitivity), 0x00, 0x00, 0x02, 0x00, lowByte(setStillSensitivity), highByte(setStillSensitivity), 0x00, 0x00};
    sendCommand(cmd, value, 18);
    queryParameters();
  }
```

crlogic, 2022-08-23:
> for individual gate sensitivity

The challenge I see here is that changing any one of those three things repeats the "set operation".  Perhaps consider only using the lambda for setDistanceGate? Which I seem to recall starting at zero.  Gotta double check that.

The other problem I see here is that without some form of reporting, this will be a nightmare to manage.   So parsing of the current setting is likely needed.

setDistanceGate might also be a good opportunity for a `select` instead of a number. But either will work.  My past attempts of using `select` always ended with far more complicated code.  So there is indeed a tradeoff.

crlogic, 2022-08-23:
> @crlogic I have not tried your code

My code is posted here --> https://community.home-assistant.io/t/mmwave-wars-the-one-sensor-module-to-rule-them-all/453260/2

webhostsg, 2022-08-23:
> https://community.home-assistant.io/t/mmwave-wars-the-one-sensor-module-to-rule-them-all/453260/2

@crlogic thank you, i'll have a look later, now finding to "sweet spot" so that the LD2410 does not detect my AC wall fan. In another room I have a DC floor fan, but it's about 4-5 meters from the sensor the "energy" is low so it does not trigger presence.
[image]

webhostsg, 2022-08-23:
Ok, found my sweet spot. The wall fan is 75cm from LD2410, set Gate 2 to 90%, and left all the rest as default (Gate 0 - 50% and Gate 1 - 50%) small movement at 6 meters was detected with no issues and does not detect motion when the fan is turned on. Setting sensitivity for the individual gates is really handy.

crlogic, 2022-08-23:
> Setting sensitivity for the individual gates is really handy.

It definitely can be.  A big sacrifice though, because the gates are so big you lose an entire ~0.75M detection segment.  In your use-case it looks ok.  Unclear how "general purpose" this makes it.

webhostsg, 2022-08-23:
@crlogic I have not tested it yet, but from the output it seems that the fan is triggering "Motion/Presence Detected" from the OUT pin from the "Rest Energy" and not motion. In my above setting i set both "Motion Sensitivity" and "Rest Sensitivity" 90%. I'll try to change the motion to the default and "Rest Sensitivity" to 90% later, time for me to sleep now.

DTTerastar, 2022-08-24:
https://twitter.com/ESPresense/status/1562439857825550336?s=20&t=q_qwe3M3k-ksdfTL4j_7Bw  I haven't tried to adjust sensitivity or use the serial so far and I'm not sure I need to it's working really well atm.  And of course my DFRobot radar sensors came hours after setting up the cheapy!

crlogic, 2022-08-24:
> And of course my DFRobot radar sensors came hours after setting up the cheapy!

So many radar!

Setting up the HLK-LD303 this week..

webhostsg, 2022-08-24:
> https://twitter.com/ESPresense/status/1562439857825550336?s=20&t=q_qwe3M3k-ksdfTL4j_7Bw I haven't tried to adjust sensitivity or use the serial so far and I'm not sure I need to it's working really well atm. And of course my DFRobot radar sensors came hours after setting up the cheapy!

Actually quite true, using the OUT pin to ESP32 GPIO works great for LD2410 too. 
It's only that one room that I need to adjust the sensitivity, I could just use the included tool if I had a Windows PC to set the sensitivity. 

But having the serial would allow not only motion detection but presence without motion. For example, if there is motion run one automation, if no motion but presence is detected run another automation.

webhostsg, 2022-08-24:
> HLK-LD303

3.5M range, maybe for a smaller room or as a bed presence sensor.

crlogic, 2022-08-24:
> 3.5M range, maybe for a smaller room or as a bed presence sensor.

I am not sure what to expect quite yet.  I have other mmwave sensor modules that are not "presence sensors" that don't report when you stand still in front of them so I am curious to see how this one behaves.

webhostsg, 2022-08-24:
@crlogic did you buy this? -> https://training.ti.com/intelligent-fall-detection-using-ti-mmwave-radar-sensors

crlogic, 2022-08-24:
> did you buy this?

$222 CDN. Lol, nope!

webhostsg, 2022-08-24:
> > did you buy this?
> 
> $222 CDN. Lol, nope!

@crlogic yup totally not affordable. Would be great if cheap ones could do fall detection.

crlogic, 2022-08-24:
SeeedStudio ?

webhostsg, 2022-08-24:
> SeeedStudio ?

@crlogic but according to your mmWave Wars post, it's too sensitive?

[13 more replies]