ESPresense/ESPresense #493: Support for atom lite

romant17, 2022-06-15:
As the atom lite is one of the recommended boards, It would be useful to add explicit support for the integrated button and rgb led.

The button is on pin 37 and can currently be recognised as a pir sensor but it would be more appropriate to explicitly recognise it as a button within mqtt/HA to be used in automations. The  rgb led is currently not functional but it would be nice to be able to set the colour of the led from within HA to use as an indicator for other states in HA.

Thanks

iCSpotRun, 2022-06-17:
Up voted!

DTTerastar, 2022-06-18:
Yes, button support is a great idea.  the m5atom-matrix firmware does control the RGB light as a status light.

cromelex, 2022-08-18:
Just to clarify, on the Firmware page there is now a "flavour" M5Atom, is this the one we should use for the AtomLite?
I have 5 of them but initially flashed the generic esp32 firmware and they update on the same firmware ever since, just wondering if I should replace it with this version of the firmware.

I see there is now a [AtomU ](https://shop.m5stack.com/products/atomu-esp32-development-kit-with-usb-a?variant=42309147132161) which to me looks like a USB-A version of the lite. This should work with the generic esp32 right?
The USB-A format with the presentable look would suit me for a couple of visible locations in my house.

DTTerastar, 2022-08-18:
Yes the M5atom supports all the M5Atoms.  This was true back when it was called m5atom-matrix.  I just renamed it to reflect the fact it can work with all of them.  I have some AtomUs and they are great little units!  I am working on generic LED support for all, so this enhancement is just to cover making the button work.

closed 2022-10-24 by stale[bot]

krazos, 2022-10-24:
To the extent this feature has not already been implemented, I would propose re-opening this issue/enhancement request.