ESPresense/ESPresense #494: Periodic abort with 2.1.2 on ESP32

jfroy, 2022-06-17:
**Describe the bug**

All my ESPresense boards (6) regularly reboot. I finally hooked one to a serial console and caught an abort.

**Config**

- Version: 2.1.2
- Active scan enabled: no
- Include filter: `apple:1007:11-7 1005:9-24 apple:1006:10-12`
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: ESP32
- Device: ACEIRMC D1 Mini NodeMCU ESP32 ESP-WROOM-32 WLAN WiFi Bluetooth IoT Development Board (https://www.amazon.com/gp/product/B08PNWB81Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

**Logs**

```
abort() was called at PC 0x401ac7fb on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f7e8:0x3ffe94d0 0x4008fa65:0x3ffe94f0 0x401ac7fb:0x3ffe9510 0x401ac842:0x3ffe9530 0x401ac16d:0x3ffe9550 0x401ac244:0x3ffe9570 0x400f82ed:0x3ffe9590 0x400f9ebb:0x3ffe95c0 0x400fa3ef:0x3ffe9620 0x400fa579:0x3ffe9690 0x400fa697:0x3ffe96b0 0x400f5d8d:0x3ffe96f0 0x400f5f9f:0x3ffe9790 0x400d30b6:0x3ffe97b0 0x400e2f19:0x3ffe97d0 0x400e55b2:0x3ffe9810 0x400e9b95:0x3ffe9880 0x400e9c41:0x3ffe98c0 0x400e9e8d:0x3ffe98e0 0x400e8c01:0x3ffe9900 0x400ef682:0x3ffe9920 0x400e2583:0x3ffe9940 0x40090a76:0x3ffe9960

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID 'XXX'.....................192.168.1.28
Firmware:     esp32
Version:      v2.1.2
IP address:   192.168.1.28
DNS address:  192.168.1.1
Hostname:     espresense-office
Room:         Office
MQTT server:  homeassistant:1883
Max Distance: 2.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      apple:1007:11-7 1005:9-24 apple:1006:10-12
Exclude:      
Known Macs:   
```

I plan to flash this board with the verbose flavor soon and will report back if it yields more info.

jfroy, 2022-06-17:
Got another crash log with a corrupt heap message:

```
CORRUPT HEAP: Bad head at 0x3fffff14. Expected 0xabba1234 got 0x3ffffff8
abort() was called at PC 0x40086d25 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f7e8:0x3ffd5540 0x4008fa65:0x3ffd5560 0x40086d25:0x3ffd5580 0x40086e51:0x3ffd55b0 0x4012714f:0x3ffd55d0 0x40120bf5:0x3ffd5890 0x40120b61:0x3ffd58e0 0x40094675:0x3ffd5910 0x40081eb6:0x3ffd5930 0x40086c1d:0x3ffd5950 0x4000bec7:0x3ffd5970 0x40132a16:0x3ffd5990 0x40137dcb:0x3ffd59b0 0x40139fa2:0x3ffd59e0 0x40090a76:0x3ffd5a10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
```

jfroy, 2022-06-17:
Caught one abort with verbose firmware.

```
Verbose | sd:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-16) BIGNUM - Memory allocation failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -16
[W][HTTPClient.cpp:1417] returnError(): error(-1): connection refused
Not updating from (sc=-1): 
abort() was called at PC 0x401ad173 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f7e8:0x3ffe1a10 0x4008fa65:0x3ffe1a30 0x401ad173:0x3ffe1a50 0x401ad1ba:0x3ffe1a70 0x401ac893:0x3ffe1a90 0x401acbca:0x3ffe1ab0 0x400dc6c1:0x3ffe1ad0 0x400daa5d:0x3ffe1b10 0x400d9021:0x3ffe1b40 0x400d9559:0x3ffe1fb0 0x40090a76:0x3ffe1fe0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
I (90) wifi:wifi driver task: 3ffd5a84, prio:23, stack:3584, core=0
I (410) wifi:wifi firmware version: dc30037
I (411) wifi:config NVS flash: enabled
I (411) wifi:config nano formating: disabled
I (411) wifi:Init data frame dynamic rx buffer num: 32
I (416) wifi:Init management frame dynamic rx buffer num: 32
I (421) wifi:Init management short buffer num: 32
I (426) wifi:Init dynamic tx buffer num: 32
I (430) wifi:Init static rx buffer size: 1600
I (434) wifi:Init static rx buffer num: 16
I (438) wifi:Init dynamic rx buffer num: 32
I (537) wifi:mode : sta (30:c6:f7:99:39:30)
Connecting to WiFi SSID '<redacted>'..............I (2604) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
.....I (3345) wifi:state: init -> auth (b0)
I (3361) wifi:state: auth -> assoc (0)
.I (3447) wifi:state: assoc -> run (10)
I (3478) wifi:connected with <redacted>, aid = 4, channel 6, BW20, bssid = <redacted>
I (3479) wifi:security type: 3, phy: bgn, rssi: -56
I (3481) wifi:pm start, type: 1

I (3489) wifi:AP's beacon interval = 102400 us, DTIM period = 1
.192.168.1.28
Firmware:     esp32-verbose
Version:      v2.1.2
IP address:   192.168.1.28
DNS address:  192.168.1.1
Hostname:     espresense-office
Room:         Office
MQTT server:  homeassistant:1883
Max Distance: 2.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      apple:1007:11-7 1005:9-24 apple:1006:10-12
Exclude:      
Known Macs:   
Count Ids:    ```

DTTerastar, 2022-06-18:
That is helpful, https://github.com/espressif/arduino-esp32/issues/2286 seems kinda related since we're hitting github via ssl to check for updates.

DTTerastar, 2022-06-18:
How often is it restarting?  minutes/hours/days?

jfroy, 2022-06-20:
> How often is it restarting? minutes/hours/days?

I just checked logs and it was sometimes several times per minute, but usually a few times per hour.

Based on the SSL logs above, I disabled auto-update and so far I've had no resets. So pretty strong indication the problem is there.

DTTerastar, 2022-06-24:
I found a bug relating to certain ble devices that transmit a name, and after that there's a chance of a corrupt heap.  SSL uses tons of memory so it's reasonable that it didn't corrupt the heap, just detected it.  Can you try the newest beta and see if it fixes the issue for you?

jfroy, 2022-06-24:
> I found a bug relating to certain ble devices that transmit a name, and after that there's a chance of a corrupt heap. SSL uses tons of memory so it's reasonable that it didn't corrupt the heap, just detected it. Can you try the newest beta and see if it fixes the issue for you?

Will do this weekend and report back.

DTTerastar, 2022-07-04:
Another user reported the latest betas fix this so closing this for now.  Please open another one w/ the beta log if you still see crashes.

closed 2022-07-04 by DTTerastar