ESPresense/ESPresense #500: Fixed Smile PRO support

hellcry37, 2022-06-25:
**Is your feature request related to a problem? Please describe.**
ID is always name:smile-pro

**Describe the solution you'd like**
Add support for device if possible:
https://www.fixed.zone/en/products/fixed-smile-pro/

**Describe alternatives you've considered**
none

**Additional context**
This tracker seems to work well after is initially paired in app, the only issue with it, I can not get an unique id from it. It is always "name:smile-pro", this is the ID with "name:" in the name

Let me know what I should send to help add this device if it is possible.

hellcry37, 2022-07-08:
this is fixed now appears as itrack:mac

closed 2022-07-08 by hellcry37

DTTerastar, 2022-07-08:
Sweet.