ESPresense/ESPresense #503: Wifi AP Enabled even after setup, jamming WIFI channels

fblgit, 2022-06-27:
**Describe the bug**

The ESP32 boards with ESPresense uses wide 2.4G Wifi AP Portal jamming the wifi channel environment.
Ideally, this AP should be disabled once the connection to the MQTT or the WIFI is established.

**Config**

- Version: v2.1.2
- Active scan enabled: no
- Include filter: yes
- Exclude filter: no
- Query filter:

**Hardware**

- Firmware flavor: ESP32
- Device (be specific): ESP32 WROOM

**Logs**

No Logs are seen, but the wifi is identified by the unifi console
[image]

**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-06-28:
I think what is happening, is you're scanning for wifi which turns off your access point, which espresense sees as losing wifi and it pops the access point to fix it, and once your wifi comes back up ESPresense has reconnected.  So I think it's working exactly as intended :)

fblgit, 2022-07-04:
No, It doesn't works like that. The interfaces are promiscuous on the unifi's, and they have multiple interfaces.. there is no disconnect to scan.. and I have tons of AP's around (expect signal redundancy).

The AP goes in when the MQTT cant be reach, but then once is reach.. the AP stills on. There should be a way to fully disable this AP by a flag or something.

Leaking the credentials of the MQTT with this AP around is pretty much simple.. on top of the missing Authentication feature of the portal..

RutgerDiehard, 2022-07-26:
I have seen Espresense captive portals from several of my ESP32s popup when checking the Unifi WiFi scan page. I've also checked and noticed that nightly optimisation was enabled.

[image]

Presumably, this performs a wireless scan across all three of my UAP-AC-Pro APs and eventually an ESP32 loses connection with MQTT and fires up its captive portal. I've disabled nightly optimisation and have not seen any portals appear since.

RutgerDiehard, 2022-07-26:
I have seen Espresense captive portals from several of my ESP32s popup when checking the Unifi Wi-Fi scan page. I've also checked and noticed that nightly optimisation was enabled.

[image]

Presumably, this performs a wireless scan across all three of my UAP-AC-Pro APs and eventually a ESP32 loses connection with MQTT during the scan and fires up its captive portal. I've disabled nightly optimisation and have not seen any portals appear since. I assume that nightly optimisation scans are the same as a specific wireless scan from an AP which does drop clients.

[image]

DTTerastar, 2022-09-02:
I think the best way of working around this is to set a timeout for connecting to wifi.  If we make it higher than the time the auto-scan takes then we won't see the AP.  If you 100% disable the AP you'll end up with a stuck node with no recovery.

DTTerastar, 2022-09-02:
I can make the timeout allow for -1 for infinity, so it'll do what you want...

PR #655 (2022-09-06): Make WiFi connection timeout configurable (Fixes #503)
Make WiFi connection timeout configurable (Fixes #503)
commit 3bb6f4c2 (2022-09-06): Make WiFi connection timeout configurable (Fixes #503) (#655)

* Make WiFi connection timeout configurable
* Pin AsyncWiFiSettings
closed 2022-09-06 by DTTerastar