ESPresense/ESPresense #507: Dallas DS18b20 sensor support

vzoltan, 2022-06-29:
Hi, could you please add DS18b20 temperature sensor too ?

it works with esp8622 if that helps, and also integrated with esphome

this is its config in ESPHome in Hme assistant

```
esphome:
  name: "easttemperature"
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.2.205
    gateway: 192.168.2.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "East temp"
    password: "Password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Password"

ota:
  password: "Password"
  
# Example configuration entry
dallas:
  pin: GPIO05
  update_interval: 30s

# Individual sensors
sensor:
  - platform: dallas
    address: 0xC2000004CDD71628
    name: "East temperature"
    resolution: 12
    accuracy_decimals: 2
```

DTTerastar, 2022-08-30:
I have one of these in my Kincaid 868-UAIR.  Will get to it at some point.

EddyMoulton, 2023-12-07:
This was added in https://github.com/ESPresense/ESPresense/pull/907 - https://github.com/ESPresense/ESPresense/releases/tag/v3.3.0

DTTerastar, 2023-12-10:
Yes it was, I forgot about that!

closed 2023-12-10 by DTTerastar