ESPresense/ESPresense #508: xiaomi_lywsd03mmc sensor support

vzoltan, 2022-06-29:
Hi, could you please add xiaomi_lywsd03mmc temperature/humidity sensor too ?

it works with esp32 if that helps, and also integrated with esphome to home assistant

this is its config in ESPHome in Hme assistant

```
substitutions:
  esphome_name: esp32tempserver

esphome:
  name: ${esphome_name}
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.2.207
    gateway: 192.168.2.1
    subnet: 255.255.255.0

logger:

api:
  reboot_timeout: 60min
  password: Password

ota:
  password: Password

time:
  - platform: homeassistant
  
esp32_ble_tracker:

sensor:

  - platform: xiaomi_lywsd03mmc
    mac_address: A4:C1:38:F1:8E:0D
    id: xiaomisensor_livingroom
    bindkey: "05e04076be48f427f3d90e166d0fbd5e"
    temperature:
      name: "Living room Temperature"
    humidity:
      name: "Living room Humidity"
    battery_level:
      name: "Living room Battery Level"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: A4:C1:38:71:06:2F
    id: xiaomisensor_bedroom
    bindkey: "05e04076be48f427f3d90e166d0fbd5e"
    temperature:
      name: "Bedroom Temperature"
    humidity:
      name: "Bedroom Humidity"
    battery_level:
      name: "Bedroom Battery Level"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: A4:C1:38:B6:A1:6D
    id: xiaomisensor_bathroom
    bindkey: "05e04076be48f427f3d90e166d0fbd5e"
    temperature:
      name: "Bathroom Temperature"
    humidity:
      name: "Bathroom Humidity"
    battery_level:
      name: "Bathroom Battery Level"
```

eric1905, 2022-07-17:
That would be great if this is working :)

DTTerastar, 2022-07-18:
I'd like to do this, but there are now like 4 ways of formatting the advertisements.  I would like to just support this format https://custom-components.github.io/ble_monitor/ha_ble If you flash it with this you can use that format: https://github.com/pvvx/ATC_MiThermometer

eric1905, 2022-07-18:
So when I flash it with the ATC firmware it is already supporting this format?

vzoltan, 2022-07-18:
this is the "original hack"

https://github.com/atc1441/ATC_MiThermometer
and im using its firmware, with "MISytle advertisement" + ESPHome as mentioned above.

DTTerastar, 2022-07-18:
Not yet.  It is on my list of stuff to do fairly high, but the passive ble monitor component works so well so far....

vzoltan, 2022-07-18:
how about integrating your project to ESPHome eventually? would be much easier to add other sensors and things to it, no?

DTTerastar, 2022-07-18:
ESPHome is really good for dumb sensors.  You can kinda get some amount of smart via lambdas.  ESPresense is really a different animal.  I want all the sensors to work together to get better presence.  People have taken parts of ESPresense and added them to ESPHome which is cool if you need the simple case.  I still have plans to have the ESPresense nodes "hive" mind to each other to refine the location of things instead of mqtt room... Just hard to get it all done late after the kids are in bed...

vzoltan, 2022-07-18:
Ok, was just asking:)

summongod, 2022-08-03:
Would it be possible to filter a single BLE device by MAC, extract the temp and humidity and then publish as a local sensor (similar to how hardware sensors are used today)?

ndbroadbent, 2022-09-27:
Hello, I can confirm that my ATC MiThermometers (`xiaomi_lywsd03mmc`) are getting picked up by ESPresense and reporting the temp/humidity data to MQTT.

You can use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html to flash the custom firmware (https://github.com/pvvx/ATC_MiThermometer), and set the Advertising Type to "ATC441".

After you disconnect from the device, you should then see the `temp` and `rh` (humidity) data showing up under a MQTT topic like `espresense/devices/miTherm:a4c138f8d65d/bedroom`:

[image]

This is really awesome, and now I don't need to set up any additional ESP32s just to run the bluetooth proxy.

But I've opened two feature requests that could make this even better:

* [Support the BTHome standard for broadcasting sensor temp/humidity data](https://github.com/ESPresense/ESPresense/issues/695)
* [Send bluetooth temp/humidity data to homeassistant topics for MQTT Discovery](https://github.com/ESPresense/ESPresense/issues/694)

In the meantime I've figured out how to set up a Node-RED function to process the MQTT messages and set up HA sensors for the closest room.

[image]

Here's the JS code for the function:

```
const originalTopic = msg.topic;
const matches = msg.topic.match(/\/miTherm:(?<id>[^/]+)\/(?<room>.+)$/);
if (!matches) return msg;
msg.device_id = matches.groups.id;
msg.room = matches.groups.room;

const distance_key = `ha_sensor_distance_${msg.device_id}`;
const room_key = `ha_sensor_room_${msg.device_id}`;
msg.distance = msg.payload.distance;
msg.existing_distance = parseFloat(global.get(distance_key) || 100000);
msg.existing_room = global.get(room_key);

const originalPayload = msg.payload;
msg.payload = [];

if (msg.distance < msg.existing_distance || true) {
    global.set(distance_key, msg.distance);

    if (msg.room != msg.existing_room || true) {
        global.set(room_key, msg.room);

        msg.qos = 2;
        msg.retain = true;
        msg.send_mqtt_message = true;
        msg.payload = [
            {
                "topic": `homeassistant/sensor/esp_bt_${msg.device_id}/temperature/config`,
                "payload": {
                    "availability": [
                        { "topic": `espresense/rooms/${msg.room}/status` }
                    ],
                    "device": {
                        "identifiers": [`espresense_bt_sensor_${msg.device_id}`], 
                        "manufacturer": "ESPresense", 
                        "model": "ATC MiThermometer",
                        "name": `ESPresense BT Sensor ${msg.device_id} Temperature`
                    }, 
                    "device_class": "temperature", 
                    "enabled_by_default": true,
                    "name": `ESPresense BT Sensor ${msg.device_id} Temperature`, 
                    "state_class": "measurement", 
                    "state_topic": originalTopic,
                    "unique_id": `${msg.device_id}_temperature_espresense`, 
                    "unit_of_measurement": "°C",
                    "value_template": "{{ value_json.temp }}" 
                }
            },
            {
                "topic": `homeassistant/sensor/esp_bt_${msg.device_id}/humidity/config`,
                "payload": {
                    "availability": [
                        { "topic": `espresense/rooms/${msg.room}/status` }
                    ],
                    "device": {
                        "identifiers": [`espresense_bt_sensor_${msg.device_id}`],
                        "manufacturer": "ESPresense",
                        "model": "ATC MiThermometer",
                        "name": `ESPresense BT Sensor ${msg.device_id} Humidity`,
                    },
                    "device_class": "humidity",
                    "enabled_by_default": true,
                    "name": `ESPresense BT Sensor ${msg.device_id} Humidity`,
                    "state_class": "measurement",
                    "state_topic": originalTopic,
                    "unique_id": `${msg.device_id}_humidity_espresense`,
                    "unit_of_measurement": "%",
                    "value_template": "{{ value_json.rh }}"
                }
            }
        ]
    }
}

return msg;
```

gerard33, 2022-10-15:
> Hello, I can confirm that my ATC MiThermometers (`xiaomi_lywsd03mmc`) are getting picked up by ESPresense and reporting the temp/humidity data to MQTT.

Nice catch @ndbroadbent. That works great and saves me the use of an extra ESP32 board with ESPHome for detecting these thermometers.

I have added my ATC MiThermometers via the [MQTT sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) integration as follows for those who want to use this and don't have NodeRed.
```yaml
mqtt:
  sensor:
    - name: ATC kantoor temperatuur
      unique_id: eb08a20e-583f-4e59-a78e-3c5ce17a39a4
      device:
        manufacturer: ESPresense
        model: ATC MiThermometer
        name: ATC MiThermometer kantoor
        configuration_url: http://192.168.1.100
        identifiers: a4c1383e33xx
        suggested_area: Kantoor
      state_topic: "espresense/devices/miTherm:a4c1383e33xx/boven"
      value_template: "{{ value_json.temp }}"
      unit_of_measurement: "°C"
      device_class: temperature
      state_class: measurement
    - name: ATC kantoor vochtigheid
      unique_id: c23966a6-d361-4554-8113-10a5713e7853
      device:
        manufacturer: ESPresense
        model: ATC MiThermometer
        name: ATC MiThermometer kantoor
        configuration_url: http://192.168.1.100
        identifiers: a4c1383e33xx
        suggested_area: Kantoor
      state_topic: "espresense/devices/miTherm:a4c1383e33xx/boven"
      value_template: "{{ value_json.rh }}"
      unit_of_measurement: "%"
      device_class: humidity
      state_class: measurement
    - name: ATC kantoor batterij
      unique_id: 8770a027-1ac2-4e40-9447-f3edd5c93c2e
      device:
        manufacturer: ESPresense
        model: ATC MiThermometer
        name: ATC MiThermometer kantoor
        configuration_url: http://192.168.1.100
        identifiers: a4c1383e33xx
        suggested_area: Kantoor
      state_topic: "espresense/devices/miTherm:a4c1383e33xx/boven"
      value_template: "{{ value_json.batt }}"
      unit_of_measurement: "%"
      device_class: battery
      state_class: measurement
```

closed 2023-01-28 by stale[bot]

jar87, 2023-12-23:
> You can use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html to flash the custom firmware (https://github.com/pvvx/ATC_MiThermometer), and set the Advertising Type to "ATC441".

Whart should i download from that link? 

https://github.com/pvvx/ATC_MiThermometer

Suschibomber, 2024-01-07:
> Hello, I can confirm that my ATC MiThermometers (`xiaomi_lywsd03mmc`) are getting picked up by ESPresense and reporting the temp/humidity data to MQTT.
> 
> You can use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html to flash the custom firmware (https://github.com/pvvx/ATC_MiThermometer), and set the Advertising Type to "ATC441".
> 
> After you disconnect from the device, you should then see the `temp` and `rh` (humidity) data showing up under a MQTT topic like `espresense/devices/miTherm:a4c138f8d65d/bedroom`:
> 
> [image]
> 
> This is really awesome, and now I don't need to set up any additional ESP32s just to run the bluetooth proxy.
> 
> I've opened two feature requests to make this better and easier to set up:
> 
> * [Support the BTHome standard for broadcasting sensor temp/humidity data](https://github.com/ESPresense/ESPresense/issues/695)
> * [Send bluetooth temp/humidity data to homeassistant topics for MQTT Discovery](https://github.com/ESPresense/ESPresense/issues/694)
> 
> In the meantime I've set up a Node-RED function to process the MQTT messages and set up HA sensors for the closest room.
> 
> [image]
> 
> Here's the JS code for the function:
> 
> ```
> const forceUpdates = false; // Set to true to update all HA entities
> 
> const originalTopic = msg.topic;
> const matches = msg.topic.match(/\/miTherm:(?<id>[^/]+)\/(?<room>.+)$/);
> if (!matches) return msg;
> msg.device_id = matches.groups.id;
> msg.room = matches.groups.room;
> 
> const distance_key = `ha_sensor_distance_${msg.device_id}`;
> const room_key = `ha_sensor_room_${msg.device_id}`;
> msg.distance = msg.payload.distance;
> msg.existing_distance = parseFloat(global.get(distance_key) || 100000);
> msg.existing_room = global.get(room_key);
> 
> const originalPayload = msg.payload;
> msg.payload = [];
> 
> if (forceUpdates || msg.distance < msg.existing_distance) {
>     global.set(distance_key, msg.distance);
> 
>     if (forceUpdates || msg.room != msg.existing_room) {
>         global.set(room_key, msg.room);
> 
>         msg.qos = 2;
>         msg.retain = true;
>         msg.send_mqtt_message = true;
>         msg.payload = [
>             {
>                 "topic": `homeassistant/sensor/esp_bt_${msg.device_id}/temperature/config`,
>                 "payload": {
>                     "availability": [
>                         { "topic": `espresense/rooms/${msg.room}/status` }
>                     ],
>                     "device": {
>                         "identifiers": [`espresense_bt_sensor_${msg.device_id}`], 
>                         "manufacturer": "ESPresense", 
>                         "model": "ATC MiThermometer",
>                         "name": `ESPresense BT Sensor ${msg.device_id}`
>                     }, 
>                     "device_class": "temperature", 
>                     "enabled_by_default": true,
>                     "name": `ESPresense BT Sensor ${msg.device_id} Temperature`, 
>                     "state_class": "measurement", 
>                     "state_topic": originalTopic,
>                     "unique_id": `${msg.device_id}_temperature_espresense`, 
>                     "unit_of_measurement": "°C",
>                     "value_template": "{{ value_json.temp }}" 
>                 }
[... 77 lines cut ...]
>                         "name": `ESPresense BT Sensor ${msg.device_id}`,
>                     },
>                     "enabled_by_default": true,
>                     "name": `ESPresense BT Sensor ${msg.device_id} Room`,
>                     "state_topic": originalTopic,
>                     "unique_id": `${msg.device_id}_room_espresense`,
>                     "value_template": msg.room // Static value for the closest room
>                 }
>             }
>         ]
>     }
> }
> 
> return msg;
> ```
> 
> This creates a device for each BT sensor, with entities for temperature, humidity, battery, and the room and distance of the nearest ESPresense base station. (The room is updated whenever it sees a smaller distance, so it will always stabilize and eventually stop changing.)
> 
> [image]

Hi, I know it's an older post but I have 8 ESPresense ESP32 devices and 12 ATC Mi thermometers and would like to use it like you do.

But since I am still at the beginning of Home Assitant it would be great to get some more information on how I have to proceed to have it just like you. 

Thank you in advance and have a happy new year :)