ESPresense/ESPresense #509: Amazfit GTR 2e shown as not_home in Home Assistant

vzoltan, 2022-06-29:
Hi, im using room-assistant on a raspberry pi, which sees and reportsd that my Amazfit GTR 2e watch is at home/room
but espresense shows as not_home.

what can be the problem?

(note: i did setup my phone, and that is shown in the given room just right with espresense, so the config is OK for the watch too.)

vzoltan, 2022-06-29:
mqtt gets this from esp32:

{"id":"known:e9b5de977bd4","idType":185,"rssi@1m":-76,"rssi":-39,"raw":0.09,"distance":0.13,"speed":0,"mac":"e9b5de977bd4","interval":2731}

But home assistant show like not_home

vzoltan, 2022-06-29:
also from the log of the esp terminal:
[13:51:16]1 New   | MAC: e9b5de977bd4, ID: known:e9b5de977bd4                                        -68dBm 
.
.
[13:56:09]1 C# +1 | MAC: e9b5de977bd4, ID: known:e9b5de977bd4                                        -92dBm (1.75m) 7ms

bu HA reports as:
[image]

while my phones BLE shows up just right.

closed 2022-06-30 by DTTerastar

vzoltan, 2022-06-30:
thank you, solved it!

nonomega, 2023-01-15:
Hello,

Sorry to bother you.
I'm using Home Assistant with ESpresence and it works like a charm using Miband 5 or 6.

But I'm not able to receive anything from my Amazfit GTR 2e even using MQTTExplorer.

I looked on device flag and saw that the GeneralDiscorevable is not active on my Amazfit GTR 2e.

 @vzoltan : as you're using the same device, could you please help me by sending me information like firmware version etc from your own device ?

Thanks in advance for your help.

nONo