ESPresense/ESPresense #510: unique_id: not allowed?

vzoltan, 2022-06-29:
[image]

could you please add unique_id support, ir it is not allowed by home assistant for mqtt_room ?

Thanks

DTTerastar, 2022-06-30:
That's totally a home assistant thing.  Most yaml configured entities don't use unique_id since you can change it in yaml.  You should request this in their repo if needed.

closed 2022-06-30 by DTTerastar

vzoltan, 2022-06-30:
ok, thanks

andrelung, 2022-10-16:
Sorry for commenting on a closed issue, I can't figure it our and hope that @vzoltan has some insight into this.

I can't get the IDs working in home assistant:
[image]

After clicking on the device:
[image]

> This entity ('sensor.es_presence_al_13') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.

Per HA-configuration it looks like I should add `entity_id` or `unique_id` but this isn't allowed by mqtt_room. What's the right way to change add my devices sensor, so that they use an unique id?