ESPresense/ESPresense #511: binary_sensor: not allowed?

vzoltan, 2022-06-29:
Getting this error:

Platform error binary_sensor.mqtt_room - No module named 'homeassistant.components.mqtt_room.binary_sensor'

of the following binary sensor:

binary_sensor:
  - platform: mqtt_room
    name: Office Mqtt Room
    sensors:
      amazfitgtr2eesp32_sticky_esp32:
        unique_id: amazfitstickyesp32
    state_topic: espresense/rooms/office/status
    json_attributes_topic: espresense/rooms/office/telemetry
    payload_on: online
    payload_off: offline
    device_class: connectivity

DTTerastar, 2022-06-30:
I see like 2 things mashed up in your config.  mqtt_room and mqtt.binary_sensor.  Those should be totally separate entities.

vzoltan, 2022-06-30:
Do you know how to add unique_id to binary_sensor that uses espresense sensor?

basically what id like to do is this:

have a second sensor based on my espresense sensor (device) that "turns off" delayed. to avoid little "errors" when the device is not seen, but im still home or in the room, to give it time to find it again. 

configuration.yaml:

```
binary_sensor:
  - platform: template
    scan_interval: 15
    sensors:
      amazfitgtr2epresence_sticky_esp:
        unique_id: amazfitstickyesp
        value_template: >-
          {{is_state('binary_sensor.amazfit_gtr_2e_esp', 'on')}}
        delay_off:
          minutes: 5
        friendly_name:
          Amazfit GTR 2e ON/OFF (ESP - Delayed)
```

esphome esp32 device config:

```
esp32_ble_tracker:
binary_sensor:

  - platform: ble_presence
    mac_address: E9:B5:DE:97:7B:D4
    id: amazfitgtr2e_esp
    name: "Amazfit GTR 2e (ESP)"

```

closed 2022-06-30 by vzoltan

vzoltan, 2022-06-30:
Solved

basically this is what i wanted to do, to have a delayed sensor for automations.

```
template:
  - binary_sensor:
      - name: Zoltans Watch presence
        unique_id: zoltans_watch_presence
        delay_off:
          minutes: 5
        state: >
          {{ not is_state("sensor.zoltans_watch", "not_home") }}
        icon: >
          {% if not is_state("sensor.zoltans_watch", "not_home") %}
            mdi:watch
          {% else %}
            mdi:watch-export
          {% endif %}
```
and now bc it has unique_id , it can be customized in home assistant to shown as presence detection sensor:
when it on 
[image]
when its off
[image]

to set to presence:

[image]

vzoltan, 2022-07-01:
(create a new automation for this)

or even without a unique_id for the original sensor that ESPresense creates, you can create a device for tracker:

```
alias: Zoltas Watch (sensor) as Tracker
description: ''
trigger:
  - platform: state
    entity_id:
      - sensor.zoltans_watch
condition: []
action:
  - service: device_tracker.see
    data:
      dev_id: amazfit
      location_name: |
        {% if states('sensor.zoltans_watch') == 'home' %}
          home
        {% elif states('sensor.zoltans_watch') == 'not_home' %}
          not_home
        {% else %}
           unknown
        {% endif %}
      source_type: router
mode: single

```