ESPresense/ESPresense #513: Xiaomi Mi Band 4 shown as not_home in Home Assistant

vzoltan, 2022-06-29:
[image]

Blue is my watch (also not_home) cyan is my MiBand (also not_home) red is my phone with Home assistand app, (in the correct room)

even the distance is correct for the two wearable.

Hi, im using room-assistant on a raspberry pi, which sees and reports that my Xiaomi Mi Band 4 is at home/room
but espresense shows as not_home.

what can be the problem?

(note: i did setup my phone, and that is shown in the given room just right with espresense, so the config is OK for the watch too.)

some lines from esp32 terminal
Band4:
[17:56:41]1 C# +1 | MAC: d0f10eab14e0, ID: known:d0f10eab14e0                                        -34dBm (0.09m) 76ms
Watch:
[17:56:41]1 C# +1 | MAC: e9b5de977bd4, ID: known:e9b5de977bd4                                        -69dBm (1.10m) 778ms
Phone
[18:05:05]1 New   | MAC: 6c14b8a132f1, ID: iBeacon:3f2fe1df-007f-4af7-b945-b1be3871308a-100-1        -92dBm 

configuration.yaml:

```
# One entry for each beacon you want to track
  - platform: mqtt_room
    device_id: "e9b5de977bd4"
    name: "Zoltans Watch"
    state_topic: "espresense/rooms"
    timeout: 10
    away_timeout: 120

  - platform: mqtt_room
    device_id: "d0f10eab14e0"
    name: "Zoltans MiBand"
    state_topic: "espresense/rooms"
    timeout: 10
    away_timeout: 120

  - platform: mqtt_room
    device_id: "iBeacon:3f2fe1df-007f-4af7-b945-b1be3871308a-100-1"
    name: "Zoltans Phone"
    state_topic: "espresense/rooms"
    timeout: 10
    away_timeout: 120
```

[image]
[image]

[image]

Any idea on this ?

DTTerastar, 2022-06-30:
The id includes the known: portion.  Put into hass the id as "known:d0f10eab14e0"

vzoltan, 2022-06-30:
Thank you for the quick replay, just before you did answer, i did figure it out :)
and it works.
Thanks again!

closed 2022-06-30 by vzoltan

DTTerastar, 2022-06-30:
Please help others by updating the documentation in a way that would have preventing you needing the help!  There is an edit button on every page of https://espresense.com

vzoltan, 2022-06-30:
ok, Ill do that

however, im at the office now, my amazfit watch is on me, and my miband is at home in the livingroom 
and home assistant shows both at the office.
(office is not in the same building, not even same street ;) )

what could be the issue here, where should i look for logs or clues?

vzoltan, 2022-06-30:
hm never mind for now, as i see by the change of the distance, its the same device bc it changes for the same value at the same time for both.
ill investigate, and let you know. thanks

vzoltan, 2022-06-30:
found it, used the same mac/id for both devices, sorry :)

ps: can you give me an exact url to what page you want me to add info to?
thanks

michaelmior, 2022-08-18:
If your Mi Band 4 is working, it would be great to add it to the [list of known working beacons](https://espresense.com/beacons).

vzoltan, 2022-08-18:
i just added to the list 
| Xiaomi Mi Band 4     | Yes      |                                             | id format: known:xxxxxxxxxxxx
| Amazfit GTR 2e        | Yes      |                                             | id format: known:xxxxxxxxxxxx

michaelmior, 2022-08-18:
@vzoltan It appears you did commit to a fork of the website repository on your branch. For changes to be reflected on espresense.com, you also need to [submit a pull request](https://github.com/ESPresense/ESPresense.github.io/compare/master...vzoltan:ESPresense.github.io:patch-2).

vzoltan, 2022-08-18:
Ok, then someone should add these two lines to that page. sorrry.