ESPresense/ESPresense #514: v2.2 crash

Mattrees007, 2022-06-30:
The ESP32 appears to boot loop with the upgrade to v2.2

[14:39:25]Setup network
[14:39:28]Connecting to WiFi SSID 'RedLeicester_Office'...................192.168.100.114
[14:39:28]Firmware:     esp32
[14:39:28]Version:      v2.2.0
[14:39:28]IP address:   192.168.100.114
[14:39:28]DNS address:  192.168.100.9
[14:39:28]Hostname:     espresense-hall
[14:39:28]Room:         hall
[14:39:28]MQTT server:  192.168.100.30:1883
[14:39:28]Max Distance: 16.00
[14:39:28]PIR Sensor:   disabled
[14:39:28]Radar Sensor: disabled
[14:39:28]Query:        
[14:39:28]Include:      
[14:39:28]Exclude:      
[14:39:28]Known Macs:   
[14:39:28]Count Ids:    
[14:39:28]1 New   | MAC: 41698b678961, ID: apple:1006:10-7                                           -49dBm 671e4f645654
[14:39:29]1 New   | MAC: 4e25954b8a33, ID: msft:cdp:0902                                             -47dBm cf0325f5179f1f90eb2720e94e6bfc77ab3fdd5f
[14:39:32]1 New   | MAC: ffff105f7b3a, ID: itag:ffff105f7b3a                                         -60dBm 
[14:39:42]1 Close | MAC: ffff105f7b3a, ID: itag:ffff105f7b3a                                         -34dBm
[14:39:52]1 Left  | MAC: ffff105f7b3a, ID: itag:ffff105f7b3a                                         -51dBm
[14:39:52]1 Close | MAC: ffff105f7b3a, ID: itag:ffff105f7b3a                                         -39dBm
[14:39:52]1 Left  | MAC: ffff105f7b3a, ID: itag:ffff105f7b3a                                         -51dBm
[14:39:57]1 New   | MAC: 54ef44380e58, ID: 54ef44380e58                                              -47dBm 
[14:40:00]1 New   | MAC: 341810797313, ID: msft:cdp:0902                                             -48dBm 316365adc64ab4e9e8f6e78e4cd4009f13c56f29
[14:40:42]ets Jun  8 2016 00:22:57
[14:40:42]
[14:40:42]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:40:42]configsip: 0, SPIWP:0xee
[14:40:42]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:40:42]mode:DIO, clock div:2
[14:40:42]load:0x3fff0018,len:4
[14:40:42]load:0x3fff001c,len:1044
[14:40:42]load:0x40078000,len:8896
[14:40:42]load:0x40080400,len:5828
[14:40:42]entry 0x400806ac
[14:40:43]Setup network
[14:40:46]Connecting to WiFi SSID 'RedLeicester_Office'...................192.168.100.114
[14:40:46]Firmware:     esp32
[14:40:46]Version:      v2.2.0
[14:40:46]IP address:   192.168.100.114
[14:40:46]DNS address:  192.168.100.9
[14:40:46]Hostname:     espresense-hall
[14:40:46]Room:         hall
[14:40:46]MQTT server:  192.168.100.30:1883
[14:40:46]Max Distance: 16.00
[14:40:46]PIR Sensor:   disabled
[14:40:46]Radar Sensor: disabled
[14:40:46]Query:        
[14:40:46]Include:      
[14:40:46]Exclude:      
[14:40:46]Known Macs:   
[14:40:46]Count Ids:    apple:
[14:40:47]1 New   | MAC: 41698b678961, ID: apple:1006:10-7                                           -52dBm 671e4f645674
[14:40:47]1 New   | MAC: 341810797313, ID: msft:cdp:0902                                             -56dBm 316365adc64ab4e9e8f6e78e4cd4009f13c56f29
[14:40:48]Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[14:40:48]Debug exception reason: Stack canary watchpoint triggered () 
[14:40:48]Core 0 register dump:
[14:40:48]PC      : 0x4008affc  PS      : 0x00000836  A0      : 0x3ffbcb30  A1      : 0x3ffbca70  
[14:40:48]A2      : 0x0000003c  A3      : 0x3ffc5980  A4      : 0x00000000  A5      : 0x3ffd3b50  
[14:40:48]A6      : 0x00000000  A7      : 0x3ffbca80  A8      : 0x8012b9d6  A9      : 0x40093644  
[14:40:48]A10     : 0x00000000  A11     : 0x00000001  A12     : 0x4008b01c  A13     : 0x3ffd3b50  
[14:40:48]A14     : 0x00000000  A15     : 0x3ffbca80  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
[14:40:48]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[14:40:48]Core 0 was running in ISR context:
[14:40:48]EPC1    : 0x4008affc  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
[14:40:48]
[... 420 lines cut ...]
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:43:39]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

DTTerastar, 2022-07-01:
```
python decoder.py -e ../.pio/build/esp32/firmware.elf -t ~/.platformio/packages/toolchain-xtensa32 -s ../backtrace.txt -p ESP32
stack:
0x4008fa84: invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
0x4008fde1: abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
0x40091cd6: vPortCPUAcquireMutexIntsDisabledInternal at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  \-> inlined by: vPortCPUAcquireMutexIntsDisabled at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
  \-> inlined by: vTaskEnterCritical at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
0x40089366: wifi_int_disable_wrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_adapter.c:192
0x400887df: coex_core_bb_reset_lock at ??:?
0x400882bb: coex_bb_reset_lock at ??:?
0x4008134b: coex_bb_reset_lock_wrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1713
0x40088079: btdm_bb_isr at intc.c:?
0x4008b081: _xt_lowint1 at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0x401dcd6b: raw_connect at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/raw.c:236
0x4012c65e: get_desc_for_int at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:236
0x400923f5: prvIdleTask at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
0x40090e2e: vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-07-01:
The above seems to indicate WiFi and Bluetooth are fighting.  I've made a stab in the dark for a fix in the 2.2.1b0 beta, can you please try that?  Do you perchance have a Cisco/Linksys WiFi router?

DTTerastar, 2022-07-01:
https://espresense.com/beta will allow you to select the beta when flashing

Mattrees007, 2022-07-01:
Thanks. I'll try the beta later. I have a tp link deco m4 mesh WiFi

Mattrees007, 2022-07-01:
Ive flashed the new beta (thank you!)

but no joy.

Using 'COM5' as serial port.
Writing at 0x0017825e... (100 %)Wrote 1479664 bytes (887190 compressed) at 0x00010000 in 20.6 seconds (effective 573.5 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[09:36:28]Setup network
[09:36:29]First contact!
[09:36:29]
[09:36:29]Starting access point for configuration portal.
[09:36:29]SSID: 'espresense-d40629'
[09:36:29]IP: 192.168.4.1
[09:37:24]2 WiFi networks found.
[09:37:42]2 WiFi networks found.
[09:37:43]ets Jul 29 2019 12:21:46
[09:37:43]
[09:37:43]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:37:43]configsip: 0, SPIWP:0xee
[09:37:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:37:43]mode:DIO, clock div:2
[09:37:43]load:0x3fff0018,len:4
[09:37:43]load:0x3fff001c,len:1044
[09:37:43]load:0x40078000,len:8896
[09:37:43]load:0x40080400,len:5828
[09:37:43]entry 0x400806ac
[09:37:44]Setup network
[09:37:47]Connecting to WiFi SSID 'RedLeicester_Office'.......................192.168.100.110
[09:37:47]Firmware:     esp32
[09:37:47]Version:      v2.2.1b0
[09:37:47]IP address:   192.168.100.110
[09:37:47]DNS address:  192.168.100.9
[09:37:47]Hostname:     espresense-test
[09:37:47]Room:         Test
[09:37:47]MQTT server:  mqtt.z13.org:1883
[09:37:47]Max Distance: 16.00
[09:37:47]PIR Sensor:   disabled
[09:37:47]Radar Sensor: disabled
[09:37:47]Query:        
[09:37:47]Include:      
[09:37:47]Exclude:      
[09:37:47]Known Macs:   
[09:37:47]Count Ids:    
[09:37:48]1 New   | MAC: 008f2760f8be, ID: msft:cdp:0902                                             -40dBm f206a9365b6b18f060c88c1c21ceadc39ea56252
[09:37:48]1 Close | MAC: 008f2760f8be, ID: msft:cdp:0902                                             -34dBm
[09:37:48]1 New   | MAC: 5b7e392b10ef, ID: apple:1006:10-7                                           -55dBm 6b1e40799807
[09:37:50]0 Reconnect timer
[09:37:51]0 Reconnecting to MQTT...
[09:37:54]0 Reconnect timer
[09:37:54]0 Reconnecting to MQTT...
[09:37:57]0 Reconnect timer
[09:37:57]0 Reconnecting to MQTT...
[09:37:57]1 New   | MAC: 416ca2b41755, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-1065-0       -65dBm 
[09:38:00]0 Reconnect timer
[09:38:00]0 Reconnecting to MQTT...
[09:38:02]1 Close | MAC: 54ef44380e58, ID: 54ef44380e58                                              -39dBm
[09:38:02]1 New   | MAC: 54ef44380e58, ID: 54ef44380e58                                              -39dBm 
[09:38:03]0 Reconnect timer
[09:38:03]0 Reconnecting to MQTT...
[09:38:06]0 Reconnect timer
[09:38:06]0 Reconnecting to MQTT...
[09:38:09]0 Reconnect timer
[09:38:09]0 Reconnecting to MQTT...
[... 174 lines cut ...]
[09:39:12]Hostname:     espresense-test
[09:39:12]Room:         Test
[09:39:12]MQTT server:  192.168.100.30:1883
[09:39:12]Max Distance: 16.00
[09:39:12]PIR Sensor:   disabled
[09:39:12]Radar Sensor: disabled
[09:39:13]Query:        
[09:39:13]Include:      
[09:39:13]Exclude:      
[09:39:13]Known Macs:   
[09:39:13]Count Ids:    apple:
[09:39:13]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[09:39:13]Debug exception reason: BREAK instr 
[09:39:13]Core 1 register dump:
[09:39:13]PC      : 0x400803c0  PS      : 0x00040236  A0      : 0x00000000  A1      : 0x800f8314  
[09:39:13]A2      : 0x3ffe86c0  A3      : 0x00000020  A4      : 0x00000003  A5      : 0x4008ad20  
[09:39:13]A6      : 0x3ffe9450  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffba934  
[09:39:13]A10     : 0x3ffc4e54  A11     : 0x00000000  A12     : 0x800f9408  A13     : 0x3ffe8770  
[09:39:13]A14     : 0x3ffe9450  A15     : 0x00000001  SAR     : 0x00000003  EXCCAUSE: 0x00000001  
[09:39:13]EXCVADDR: 0xffffffe0  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[09:39:13]
[09:39:13]ELF file SHA256: 0000000000000000
[09:39:13]
[09:39:13]Backtrace: 0x400803bd:0x800f8314Guru Meditation Error: Core  1 panic'ed (Double exception)
[09:39:13]Core 1 register dump:

DTTerastar, 2022-07-02:
Please check if v2.2.1b1 helps.

DTTerastar, 2022-07-02:
Make that v2.2.1b2, i messed up making v2.2.1b1

jscherry, 2022-07-02:
I have the boot loop problem still on Version: v2.2.1b2. Using Unifi AP's, switches, a UDM Pro

DTTerastar, 2022-07-02:
Backtraces would be helpful if you can add them

Mattrees007, 2022-07-02:
Sorry. What’s one of them?

Thanks

Matt

DTTerastar, 2022-07-02:
You'll see them on the serial console they look like:
```
Backtrace: 0x4008fa84:0x3ffd5380 0x4008fde1:0x3ffd53a0 0x40091cd6:0x3ffd53c0 0x4008936f:0x3ffd53f0 0x40088917:0x3ffd5410 0x4008842d:0x3ffd5440 0x400884ab:0x3ffd5460 0x400886ad:0x3ffd5480 0x401299e2:0x3ffd54c0 0x4012b2c9:0x3ffd54e0 0x4018c7a6:0x3ffd5500 0x4018c833:0x3ffd5520 0x4018d0e2:0x3ffd5540 0x401914cd:0x3ffd5580 0x40191541:0x3ffd55a0 0x40190c21:0x3ffd55c0 0x40090e2e:0x3ffd55f0
```

Mattrees007, 2022-07-04:
v2.2.1b2 seems stable for me. its been running for 30mins with no restarts.

commit eb8d4916 (2022-07-02): Use tasmota platform, fix chipid, fixes #514? (#517)
closed 2022-07-04 by DTTerastar