ESPresense/ESPresense #522: 3 devices combined to form 1 device/dashboard and 1 entry for mqtt

Anto79-ops, 2022-07-04:
Hi!

I'm running HAOS Home Assistant 2022.7.0b0 and also, have 4 ESP32 running 2.2.1b3 on 3 of them and 2.2.1 on 1 of them. For whatever reason, the 3 on 2.2.1b3 have combined into 1 device via mqtt and shown here:

[image]

not sure how or why this happened. 

MQTT explorer is reporting 4 different devices in homeassisstant/sensors but its all mashed as one device (3 sensors) and 1 in the UI

this is a strange one to explain be happy to give more informatoin

DTTerastar, 2022-07-04:
Yes, I just noticed this as well.  I've fixed it in b4.  Coming in 20 minutes...

Anto79-ops, 2022-07-04:
Awesome, thanks for your efforts here. After it updates, how do I actually fix the combination of entities? Do I need to purge the mqtt entries (using mqtt explorer)?

Cheers

DTTerastar, 2022-07-04:
Mine are updating, trying a reboot of hass now.  I think that is all that should be needed.

Anto79-ops, 2022-07-04:
thank you! problem solved. Cheers

closed 2022-07-04 by Anto79-ops