ESPresense/ESPresense #533: Update Button

VACIndustries, 2022-07-08:
Requesting a feature to the UI where you can force the update to the latest (or beta if you're checked for it) via an "update" button instead of waiting for it to happen automatically.

DTTerastar, 2022-07-08:
Upvote this: https://community.home-assistant.io/t/mqtt-discovery-for-the-new-update-sensor/427146

closed 2022-08-23 by DTTerastar

blacknell, 2022-08-26:
Is there an MQTT topic so force an upgrade?

DTTerastar, 2022-08-28:
yes /update/set with any value sent to it

DTTerastar, 2022-08-30:
In v3 you can send a http/https url and it'll install that exact version.  Can be used to switch firmware flavors without serial!

Powerdata44, 2024-08-08:
> yes /update/set with any value sent to it

what value?

N3rdix, 2024-09-05:
> > yes /update/set with any value sent to it
> 
> what value?

as mentioned **any** value is possible, i just sent a raw `1` to the topic and the update went through

Powerdata44, 2024-09-05:
Am trying to understand on it.

At moment esp32 with espresense firmware hidden above ceiling in centre of room, connected to WiFi. Old firmware at moment.

Like to upgrade firmware to newer - can it be done through MQTT with typed as you mentioned  /update/set with add 1 in raw then will upgrade newer firmware.

Am I correct? 

Or need to do with OTA - if so how?

Where get url / https  or just

N3rdix, 2024-09-06:
if you are on the version which introduced https://github.com/ESPresense/ESPresense/pull/595 then of course you can update via MQTT. Otherwise why not accessing the local UI via `http://<IP>` and either select "update automatically" or enter the update URL (which should be a bin file from the release, e.g. https://github.com/ESPresense/ESPresense/releases/tag/v3.3.2)

Via MQTT here's my example topic which triggered the update. Hope that makes it clear.

[image]

Powerdata44, 2024-10-07:
Now I can see new update version v3.3.5 so where can I get url from in github site?

So I can add it in setting in esp32 device (hidden above ceiling mean need update via OTA.

N3rdix, 2024-10-08:
Depends on your device, the ota files are always linked in the release, e.g.:

https://github.com/ESPresense/ESPresense/releases/download/v3.3.5/esp32.bin

commit 07228feb (2022-08-23): Selectable LEDs and Update Button (#595)

* RGB LEDs supported (closes #297) 
* Add button to manually update (closes #533)
* Manual Wifi setup via web installer (closes #593, #172)
* Add esp32c3-verbose flavor (#134)
* Make scan all wifi the default (#594)
* Improvements to M5Atom/M5StickCs