ESPresense/ESPresense #536: No BME280 I2C Sensor data received via MQTT after the autoupdate to version 2.2.3

ziolelle, 2022-07-08:
BME280 i2C Sensor not available anymore from Home Assistant.

Firmware 2.2.3 (auto-updated today)

**Hardware**
ESP32 DevKitV1

Today after the auto-update to 2.2.3 without touching anything no data is received any longer from I2C sensor BME280 (at 0x76) in my hgome assistant (via MQTT).

I saw also that the old sensor names in HomeAssistant are no longer valid and I found 3 new names:
BME Umidity
BME Pressure
BME Temperature

But all of them I get 'Unknown' status

Looking with MQTT Explorer no data regarding temperature/pressure/humidity is sent.

I repeat, I didn't changed anything (concerning the hardware) that before the autoupdate was working.

Moreover, after the autoupdate, I had to physically reboot the devices because they were unreachable by Home Assistant.

FsFalcon, 2022-07-08:
I'm seeing the same issue; I ended up rolling back to v2.2.0 and turning off auto-updates to resolve it (for now).

DTTerastar, 2022-07-08:
Can you check the discovery topics in mqtt to find the difference?  I redid all the mqtt discovery so I must have broken something.

ziolelle, 2022-07-08:
in main.cpp line 586 the Setup for BME280 is commented. Is there a particular reason for this ?
thanks
Daniele

DTTerastar, 2022-07-08:
Oh, that's totally it.  I moved the setup into that method, previously it didn't use it.

DTTerastar, 2022-07-08:
v2.2.4 is building now, will auto update to it after 30mins+-10mins

ziolelle, 2022-07-08:
That's great, really thank you !!!!!

ziolelle, 2022-07-08:
Sorry but the fix is not complete..
in the autoconfig topics:

homeassistant/sensor/espresense_xxxxxx/bme_temperature/config
homeassistant/sensor/espresense_xxxxxx/bme_humidity/config
homeassistant/sensor/espresense_xxxxxx/bme_pressure/config

the part that tells to Home Assistant which topic is the value is set to:
"stat_t": ""~/bme_temperature"
"stat_t": ""~/bme_humidity"
"stat_t": ""~/bme_pressure"

but the real data topic received by HA related to temperature/humidity/pressure are:
espresense/rooms/roomname/bme280_temperature
espresense/rooms/roomname/bme280_humidity
espresense/rooms/roomname/bme280_pressure

I think they must be fixed correspondly in order to be "autodiscovered" by Home Assistant.

ciao
Daniele

DTTerastar, 2022-07-08:
v2.2.5 has the fix, let me know!

ziolelle, 2022-07-09:
Last fix in v2.2.5 works perfectly!
Thank you !!!
Daniele

FsFalcon, 2022-07-09:
@ziolelle @DTTerastar  I'm still seeing the same issues after updating to v2.2.5; do the MQTT auto-discovery topics have to be purged in order for Home Assistant to start showing the updated values?

ziolelle, 2022-07-09:
@FsFalcon using a program like MQTT Discover, I deleted all the stuff regarding the autodiscovery for HA related to ESPresence.
Restarted the ESPresence beacons and with the MQTT autodiscovery active in HA everything was clean and fine.

ziolelle, 2022-07-09:
as   #540 again no data retrieved from BME280 i2C
Even enabling I2C scanning no activity present on Web Terminal

DTTerastar, 2022-07-09:
v2.2.6 should fix it.

ziolelle, 2022-07-09:
v2.2.6 is working !
Thank you

closed 2022-07-09 by ziolelle

FsFalcon, 2022-07-11:
> @FsFalcon using a program like MQTT Discover, I deleted all the stuff regarding the autodiscovery for HA related to ESPresence. Restarted the ESPresence beacons and with the MQTT autodiscovery active in HA everything was clean and fine.

@ziolelle - Closing the loop here; I purged the auto-discovery topics from HA (using MQTT Explorer) and everything is now working as expected under v2.2.7.  Thanks!