ESPresense/ESPresense #540: v2.2.5 Broke BH1750 sensor

simonyel, 2022-07-09:
**3 hours ago my devices updated to v2.2.5 and since then all rooms read static lux from BH1750**

**Config**

- Version: v2.2.5
- Active scan enabled: No
- Include filter: iBeacon: garmin:

**Hardware** ESP32-DevKit

- Firmware flavor: ESPresense Version: v2.2.5
- Device (be specific): Homebrew: ESP32-DevKit with PIR, DHT22, BH1750

simonyel, 2022-07-09:
[image]
[image]
[image]
[image]

ziolelle, 2022-07-09:
I can confirm that also the BME280 Sensor data is not acgtive any longer.
LAst night after first push of data no more data has been received. 
Deleting vcia MQTT Explorer the retained data received once, no more data has been received.

ziolelle, 2022-07-09:
in order to givwe more details, enabling the I2C Scanning, no scan is started looking via WebTerminal.
The sensor is found correctly, but no scanning activity is done:
```
Connecting to WiFi SSID 'xxxxxxxxxxxx'................192.168.1.202
Firmware:     esp32-sensors
Version:      v2.2.5
IP address:   192.168.1.202
DNS address:  192.168.1.254
Hostname:     espresense-xxxxxxxxx
Room:         Studio
MQTT server:  192.168.1.xxx:1883
Max Distance: 3.00
PIR Sensor:   disabled
Radar Sensor: disabled
DHT11 Sensor: disabled
DHT22 Sensor: disabled
DHT Temp Offset: disabled
BH1750_I2c Sensor:  on bus 1
BME280_I2c Sensor: 0x76 on bus 1
TSL2561_I2c Sensor:  on bus 1
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:  
```

DTTerastar, 2022-07-09:
v2.2.6 should fix it.

simonyel, 2022-07-09:
2.2.6 has BH1750 working again. 

- Thank you!

closed 2022-07-09 by simonyel