ESPresense/ESPresense #545: Completely remove espresense from ESP32

pickonedev, 2022-07-12:
Hi!

Can someone please tell me how can I wipe ESP32 completely? Like a factory reset...
I tried with `py -m esptool --chip ESP32 --port COM5 erase_flash` but espresense it is still there...

Thank you in advance!

hellcry37, 2022-07-13:
Write something else over it? Use ESP32 for whatever else you want! I'am no expert but I don't think you can factory clean esp32 chip.

pickonedev, 2022-07-14:
> Write something else over it? Use ESP32 for whatever else you want! I'am no expert but I don't think you can factory clean esp32 chip.

I tried, but no result...

pickonedev, 2022-07-14:
> https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html that says erase_flash should have worked. Do you have more than one connected to usb perhaps?

No, only one used... I tried to install Ultrasonic Sensor with ESPHome from home assistant, and you can see that the ESPresense it is still there...  Take a look at the image
[image]

FsFalcon, 2022-07-14:
@pickonedev - If the issue is that you no longer want the device in Home Assistant, you'll need to remove it by purging the MQTT auto-discovery topics from your MQTT server.  This can be done with a program like [MQTT Explorer](http://mqtt-explorer.com/).

Once connected to your broker, purge any of the topics associated with the device from the homeassistant/[sensor_type]/topics  You can find the appropriate entries to purge from the 'MQTT Info' option on the device in HA.
[image]

pickonedev, 2022-07-14:
Uuuuuuu, nice, Thank you!

closed 2022-07-14 by pickonedev