ESPresense/ESPresense #546: EPS32 Disconnecting after a couple of hours

Bruhildo, 2022-07-14:
I have been using ESPpresence with no issues for some time, but recently (about 4 days ago) each Base Station (I have 2) have disconnected and stop reporting beacons. Base Stations are still connected to WiFi and MQTT but no longer reporting more updates.

MQTT logs do not show anything unusual.

If connected to serial monitor, when issue happens it reports **"scan_evt timeout"** over and over.

Both Base Stations are Generic ESP32 DevBoards and issue is happening with both boards. When issue happens, status light (blue) stops flashing. Red light (power) stills on.

Usually a soft reset (via HA interface) or a hard reset (unplugging or ESP32 Board physical button) solves the issues for another couple of hours.

51av0sh, 2022-07-14:
Ditto. This started with 2.2.7.  I've had to restart all my ESPresense base stations four times the last 3 days. Never had this issue with previous versions.

Bruhildo, 2022-07-14:
I noticed on 2.2.6 but still persists on 2.2.7. I am not sure if it happened before because latest 4 or 5 updates were done in quick succession.

maromaro70, 2022-07-16:
Yep, 5 nodes, all with terrible reporting within a few hours of restart. Started around 5-6 days ago

ASNNetworks, 2022-07-16:
Same here, some devices simply don't get listed anymore even after restarts now. Sad after every few updates related bugs like this come back. I think I'm gonna downgrade to a much older version which was rock solid and turn off auto update.

uspino2, 2022-07-16:
Happening to me too. Solid for months and it just bricks after a few hours.

uspino2, 2022-07-18:
Turns out it was -in my case- a WiFi issue in my UniFi home network. The ESP32 device insisted on connecting to a far away AP, even though it's literally 2 ft away from another AP. I locked it to that AP and it's been rock solid for the last 48 hours.

DTTerastar, 2022-07-18:
These last set of updates did update the core Arduino framework, which could cause issues.  If you have a log of scan_evt timeout it would be helpful.  My own testing is actually seeing uptimes in 600,000 seconds which we've never even seen before: 
[image]

I am wondering if such long uptimes are causing other bugs to surface...

Bruhildo, 2022-07-18:
Hi @DTTerastar , thanks for the response.

When connected at the terminal via USB, the only info on the logs was "scan_evt timeout" over and over. Before that, logs seems normal, but I could not capture the exact moment when went from normal to "scan evt timeout" as I could not go back on the terminal up to the moment it happened. Is there a different way to capture the logs?

As per uptime, looking at history, usually it stops working at 40000 (average) but this I think reflects when I noticed and reset it. But I also see that I reset quite often at around 15000, so that may reflect better when it stops working.

DTTerastar, 2022-07-18:
Seems related: https://github.com/espressif/arduino-esp32/issues/5860

Bruhildo, 2022-07-18:
@DTTerastar I tried again looking at the logs and just drop directly from normal operation to "scan_evt timeout", no other messages (sorry, I was not able to quickly grab a screen shot before it disappeared). 

Besides the beacons reports (New, Left, Close, Del), the only different message on web terminal logs was:
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.2.7/esp32.bin

DTTerastar, 2022-07-18:
I think this is an issue when there's crazy amounts of advertisements.  Do you like somewhere high density with lots of BLE around you?

Bruhildo, 2022-07-18:
I just enable "prerelease" on one of the base stations and left the other normal version (to compare if needed).

Thanks a lot for that.

Bruhildo, 2022-07-19:
As feedback, I got prerelease version 2.2.8 yesterday and still stopping after a few hours and reporting "scan_evt timeout".

DTTerastar, 2022-07-19:
OK.  I'm going to investigate a way to use watchdog to at least work around this until we can figure it out.

Bruhildo, 2022-07-25:
After some digging this weekend and review of the configuration of the base stations, I noticed that “Request scan results” was turned on. After I turned off there was no more drops.

This was not an issue before latest updates (since it was likely checked on since I first installed without issues), but turning off seems to have resolved the issue. I am without issues for 2 days already.

mwhdc, 2022-07-25:
No logs handy, but anecdotally I've been seeing this behavior too, on multiple M5Stack Atom Lites with both the vanilla and sensors builds of 2.2.7. After about a day of uptime, the ESP32s remain pingable and reachable via HTTP, and continue to report telemetry, but stop reporting devices. The issue isn't 100% reproducible, though - one of my ESP32s has been reporting like a champ for over 5 days.

DTTerastar, 2022-08-04:
3.0b1 has a work around for this.  If the known devices drops to zero we reboot.  I'm pretty sure it's an issue in the Arduino framework and i am also testing the newest version to see if it fixes it.

Bruhildo, 2022-08-05:
Thanks @DTTerastar . V3.0.0b0 was dropping about once a day (without “Request scan results” turned on). 

I have not seen a drop on V3.0b1 since you pre-released it.

Buckeyes1995, 2022-08-11:
I am seeing something similar with the latest beta.  When I get up in the morning, the com port is locked and I have to restart the ESP32 to get it going again.  Any thoughts?  I left it connected to putty this morning so will report if I see anything prior to it happening again.

DTTerastar, 2022-08-12:
Do you have request scan response turned on?

Buckeyes1995, 2022-08-12:
I do not.  Woke up this morning and had to reboot.

DTTerastar, 2022-08-30:
v3 is out and I think this is resolved, can you confirm?

Buckeyes1995, 2022-08-30:
Oh sorry, I haven’t noticed this recently.  All good.

closed 2022-08-30 by DTTerastar

sami1369, 2023-01-12:
> After some digging this weekend and review of the configuration of the base stations, I noticed that “Request scan results” was turned on. After I turned off there was no more drops.
>
> This was not [...]

Hi Thank, 
I have same problem 

Where is “Request scan results” ? 

would you elaborate on more details?

Bruhildo, 2023-01-14:
> > After some digging this weekend and review of the configuration of the base stations, I noticed that “Request scan results” was turned on. After I turned off there was no more drops.
> > This was not an issue before latest updates (since it was likely checked on since I first installed without issues), but turning off seems to have resolved the issue. I am without issues for 2 days already.
> 
> Hi, I have the same problem
> 
> Where is “Request scan results”?
> 
> would you elaborate on more details?

@sami1369 This was on an older version. As I see, this option is no longer available. Also, the issue I was facing was resolved after a new version was pushed. Maybe your issue is caused by something different than the one I had.

sami1369, 2023-01-14:
> @sami1369 This was on an older version. As I see, this option is no longer available. Also, the issue I was facing was resolved after a new version was pushed. Maybe your issue is caused by [...]

What version did you use?
I still have this problem?