ESPresense/ESPresense #550: Iphone 13 seems to now have issues with identification and location

begunfx, 2022-07-15:
**Describe the bug**

We only have one IPhone (13) in the house (the other two phones are Android).  The Android phones using a beacon app work pretty much all the time.  Until a few months ago the IPhone was also working well.  Now, it seems like the ESPLite sensors are having a hard time keeping tabs on where it is or if it's there at all.   I'm not sure what has changed, because it was working great for a while.  I haven't changed anything regarding the sensors, software settings or otherwise. 

**Config**

- Version: v2.2.7
- Active scan enabled: Unknown
- Include filter: iBeacon:1fb27f22-e037-4d7a-8644-ed01b4b99ab8-19-73 iBeacon:5e8fbd77-f32d-460b-8ee1-9be5497a1e26-19-81 apple:iphone13-2
- Exclude filter:
- Query filter:

**Hardware**

- Firmware flavor: ESP32 (Atom Lite)
- Device (be specific): Atom Lite

**Logs**

If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-09-01:
I found the issue. Please try updating to the alpha via http://espresense.com/alpha

DTTerastar, 2022-09-01:
Released in v3.0.4

closed 2022-09-01 by DTTerastar

begunfx, 2022-09-02:
So far so good.  Thanks Darrell!

DTTerastar, 2022-09-02:
Awesome!

begunfx, 2022-09-21:
@DTTerastar seems like we have a regression with this issue.  I'm now not getting any updates from my son's IPhone13.   It's always listed as away.