ESPresense/ESPresense #551: Problems connecting to Nest-Wifi netrork

bogorad, 2022-07-17:
**Describe the bug**

I have a Nest-Wifi network, one Nest master, and two google-wifi slaves. By (Google's) design, they all have the same SSID. I have many m5stack-atom-lite devices in different rooms. I set them up in my office, then carry them to the destination room. Most work great, but some just won't connect to wifi. But when I return them to my office, they work fine. It looks like ESPresence stores the MAC address and not the SSID. Which is a problem in the case of Nest/Google-wifi. Can you at least show MAC addresses in the network setup drop-down?

**Config**

- Version: v2.2.7
- Active scan enabled: no
- Include filter: yes
- Exclude filter: no
- Query filter: no

**Hardware**

- Firmware flavor: ??
- Device (be specific): M5stack Atom Lite

onlyoneme, 2022-07-17:
I have access point and an extender with the same SSID and different BSSIDs derived from MACs. My espresense devices have no issue connecting any of them, so I guess there must be another reason for your wifi problems.

DTTerastar, 2022-07-18:
I specifically don't store the BSS in nvs so the station shouldn't get sticky: https://github.com/ESPresense/ESPresense/blob/b298730daade3ee6d7de1ee053e3b5b42af41245/src/main.cpp#L137

If you used old firmware before I added that it's possible you still have an old value in your nvs.  If you check the "erase" option and reflash it should blow it away.

jeffreyameyer, 2022-08-08:
I believe this may be an issue with the esp wifi drivers, but I'm having the same issue with Google Wifi. Info below & cross-posted at:

Hi - adding to this thread here with an additional use case: trying to get ESPresence working with a Google WiFi network (no control over 2.4GHz / 5GHz bands, multiple 2.4GHz BSSIDs). In this environment, if I successfully set up a device in my office and then try to move it to another room, there's a *very* small chance it will wake up and associate properly. If I instead go to a physical location where I plan to deploy an ESP32 sensor and then reset the wireless from there, I can improve those odds, but it would be nice to be able to just give an SSID and have the ESP32 manage across all the different BSSIDs associated with that SSID.

My ESP: AITRIP 5PCS ESP-WROOM-32 ESP32 ESP-32S Type-C USB

Here's a little more info on my network and my setup:
[image]

[image]

[image]

Diddlik, 2022-08-11:
I have the same problem with Unifi Access point. it is trying to connect and break with this error message:
[image]
If I do the WiFi scan, than I have the same picture as [jeffreyameyer](https://github.com/jeffreyameyer)

DTTerastar, 2022-08-12:
I have unifi and it's working for me.

onlyoneme, 2022-08-12:
Maybe there are roaming issues there. Or esp device doesn't try to connect to another AP with the same SSID when the preferred (with strongest signal) one denies to connect on startup.

Diddlik, 2022-08-13:
> I have unifi and it's working for me.

do you have multiple AP? I have 4 of them and under wifi scan epresence shows me 4 time my ssid

DTTerastar, 2022-08-13:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#esp32-wi-fi-station-connecting-when-multiple-aps-are-found

I see mention of WIFI_ALL_CHANNEL_SCAN vs WIFI_FAST_SCAN.  Maybe that's the option we want?

DTTerastar, 2022-08-13:
https://github.com/espressif/arduino-esp32/blob/cf01523ded6d40e6feaa92b9aa647e5553148b60/libraries/WiFi/src/WiFiSTA.cpp it looks to me like WIFI_ALL_CHANNEL_SCAN is the default already.

DTTerastar, 2022-08-13:
Nm: https://github.com/espressif/arduino-esp32/blob/cf01523ded6d40e6feaa92b9aa647e5553148b60/libraries/WiFi/src/WiFiSTA.h#L69

closed 2022-08-13 by DTTerastar

Diddlik, 2022-08-13:
@DTTerastar that means in the next release it should be fixed?

jeffreyameyer, 2022-08-14:
@DTTerastar - thank you! Looking forward to the update.

Diddlik, 2022-08-15:
@DTTerastar Just tested new beta (3.0b6), for me the problem still exist

DTTerastar, 2022-08-15:
> @DTTerastar Just tested new beta (3.0b6), for me the problem still exist

I can only do what the apis provided by Esspressif allow.  Not sure what else I can do.

Diddlik, 2022-08-19:
> > @DTTerastar Just tested new beta (3.0b6), for me the problem still exist
> 
> I can only do what the apis provided by Esspressif allow. Not sure what else I can do.

I found the problem. If I remove special charackters from WiFI password, for example _ or ! than I can connect to WiFi.

DTTerastar, 2022-08-22:
A bug about exactly what char is causing issues would be awesome so at a minimum we have it documented

Diddlik, 2022-08-23:
I will do in next days more tests and than create an issue

commit 76d31895 (2022-08-13): Add options to Wifi to scan all APs (Fixes #551)
commit 1de582e7 (2022-08-13): Add options to Wifi to scan all APs (Fixes #551)
commit 83d0250e (2022-08-13): Add options to Wifi to scan all APs (Fixes #551) (#588)
commit c0f1939e (2022-08-13): Add options to Wifi to scan all APs (Fixes #551)