ESPresense/ESPresense #552: Home assistant configuration.yaml example config block is confusing for yaml rookies

UltraSalem, 2022-07-18:
**Is your feature request related to a problem? Please describe.**
As a yaml rookie, I found the example home assistant configuration yaml block to be confusing when trying to add a second beacon device. Doe to the location of the code comment, I was unsure what parts get repeated for the second beacon or not.

**Describe the solution you'd like**
I submitted a pull request onto the upstream mqtt_home where that part of the doco was generated originally, but they declined it as their project is closed - https://github.com/jptrsn/ESP32-mqtt-room/pull/57

Suggested block:
```
# One sensor entry for each beacon you want to track
sensor:
  - platform: mqtt_room
    device_id: "fda50693a4e24fb1afcfc6eb07647825-5-0" # Note that major version must match (the second last digit - '5' in this example), but any minor version will be ignored (the final digit, '0' in this example)
    name: "iBeacon Room Presence"
    state_topic: "room_presence"
    timeout: 60
    away_timeout: 120

sensor:
  - platform: mqtt_room
    device_id: "a1...

```

DTTerastar, 2022-07-18:
I didn't even know you could put multiple sensors.  This syntax is now actually out of date.  It should be under a mqtt: in the latest hass, right?

UltraSalem, 2022-07-18:
I'm using the syntax as above and it's working for me! The documentation should probably also mention doing a ctrl-f5 on the home assistant browser window after editing the configuration and restarting. I had to do that to get changes to appear, restarting home assistant alone wasn't enough.

hellcry37, 2022-07-19:
> 

you are right, it should be like this:

Under sensors:
```yaml
sensors:
  # ESPresense Room Sensors
  - platform: mqtt_room
    device_id: "iTrack:xxxxxxxxxxxx"
    name: 'ESP Keys'
    state_topic: 'espresense/devices/iTrack:xxxxxxxxxxxx'
    timeout: 10
    away_timeout: 120
  - platform: mqtt_room
    device_id: "iBeacon:xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx-3-58241"
    name: 'ESP Chipolo'
    state_topic: 'espresense/devices/iBeacon:xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx-3-58241'
    timeout: 10
    away_timeout: 120
```

Under mqtt:
```yaml
mqtt:
  binary_sensor:
    - unique_id: 'kitchen'
      name: 'Kitchen'
      state_topic: espresense/rooms/kitchen/status
      json_attributes_topic: espresense/rooms/kitchen/telemetry
      payload_on: online
      payload_off: offline
      device_class: connectivity
    - unique_id: 'living_room'
      name: 'Living Room'
      state_topic: espresense/rooms/living_room/status
      json_attributes_topic: espresense/rooms/living_room/telemetry
      payload_on: online
      payload_off: offline
      device_class: connectivity
    - unique_id: 'kid_room'
      name: 'Kid Room'
      state_topic: espresense/rooms/kid_room/status
      json_attributes_topic: espresense/rooms/kid_room/telemetry
      payload_on: online
      payload_off: offline
      device_class: connectivity
    - unique_id: 'master_bedroom'
      name: 'Master Bedroom'
      state_topic: espresense/rooms/master_bedroom/status
      json_attributes_topic: espresense/rooms/master_bedroom/telemetry
      payload_on: online
      payload_off: offline
      device_class: connectivity
```

onlyoneme, 2022-07-19:
Different setups and different behaviors above. Mqtt doesn't equal mqtt_room.

DTTerastar, 2022-07-21:
Nice.  You can PR this directly by pressing the button on the bottom of the docs that say [Edit this page on GitHub](https://github.com/ESPresense/ESPresense.github.io/edit/master/configuration/home_assistant.md)

hellcry37, 2022-07-21:
Done, I've edited the page, you can merge if It's OK

DTTerastar, 2022-07-21:
✅

closed 2022-07-21 by DTTerastar

DTTerastar, 2022-07-21:
That whole page needs work.  The bash script on the bottom can be replaced by sending a message to the * room instead.