ESPresense/ESPresense #559: Reboot loop after flashing the firmware to ESP32

admbeni, 2022-07-19:
My ESP32 board goes into reboot loop after flashing the firmware.
I tried the update from Google Chrome and with ESPHome flasher also.

Chip info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S0WD (revision 1)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES

Log content:
[08:12:53]Rebooting...
[08:12:53]ets Jun  8 2016 00:22:57
[08:12:53]
[08:12:53]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:12:53]configsip: 0, SPIWP:0xee
[08:12:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:12:53]mode:DIO, clock div:2
[08:12:53]load:0x3fff0018,len:4
[08:12:53]load:0x3fff001c,len:1044
[08:12:53]load:0x40078000,len:8896
[08:12:53]load:0x40080400,len:5828
[08:12:53]entry 0x400806ac
[08:12:54]
[08:12:54]abort() was called at PC 0x400827f0 on core 0
[08:12:54]
[08:12:54]
[08:12:54]Backtrace:0x400831c9:0x3ffe3b500x4009534d:0x3ffe3b70 0x4009a2ad:0x3ffe3b90 0x400827f0:0x3ffe3c10 0x40079053:0x3ffe3c40  |<-CORRUPTED
[08:12:54]
[08:12:54]
[08:12:54]
[08:12:54]
[08:12:54]ELF file SHA256: 0000000000000000
[08:12:54]
[08:12:54]Rebooting...

DTTerastar, 2022-07-19:
ESP32-S0WD is not currently supported only ESP32-D0WD

admbeni, 2022-07-20:
Thank you for the answer, is it possible to support ESP32-S0WD in the not
so far future?
Currently I have only this kind of chip and it would be a huge help for me.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mentes
a vírusoktól. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Darrell ***@***.***> ezt írta (időpont: 2022. júl. 20., Sze,
0:36):

DTTerastar, 2022-07-20:
I'm only one dev and with D0WDs being less than 5 bucks it's not a huge priority. So it won't happen fast.  I do believe some Sonoff's use S0WD so it is a useful addition.

closed 2022-09-26 by stale[bot]