ESPresense/ESPresense #562: Allow bluetooth pairing for espresence

thatnokiaguy, 2022-07-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Wear os does not offer any beacon i know of. solution to this would be to allow people to pair a device directly to espresence. would also possibly help to make it more accurate as the device may be more ikely to respond
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
allow us to pair bluetooth devices to the board
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
i sideloaded a beacon app onto my watch, which failed miseravly
**Additional context**
Add any other context or screenshots about the feature request here.

DTTerastar, 2022-07-21:
The prerelease/beta has this.  Press Enroll and ESPresense looks like a Heart Rate Monitor long enough to pair with.

thatnokiaguy, 2022-07-21:
Ah. Thanks for letting me know

thatnokiaguy, 2022-07-21:
> The prerelease/beta has this. Press Enroll and ESPresense looks like a Heart Rate Monitor long enough to pair with.

im unable to even get it to update to beta

thatnokiaguy, 2022-07-21:
> > The prerelease/beta has this. Press Enroll and ESPresense looks like a Heart Rate Monitor long enough to pair with.
> 
> im unable to even get it to update to beta

ok, managed to update it however cannot find enroll button, any idea?

PuckStar, 2022-07-22:
I'm on version 2.2.8b2 but when scanning for BT devices on my phone I don't see anything that looks like Espresense.

ASNNetworks, 2022-07-24:
> The prerelease/beta has this. Press Enroll and ESPresense looks like a Heart Rate Monitor long enough to pair with.

Can you explain how this works? I tried pairing (I could find the device) but after adding my phone says it needs an app in order to connect. Also, with having 4 base stations, should I pair to each of them?

linglingpin, 2022-07-24:
If you're using Home Assistant, the Enroll button is on the ESPresense device page.

ASNNetworks, 2022-07-24:
> If you're using Home Assistant, the Enroll button is on the ESPresense device page.

Not sure if your comment was towards me, but as I stated: I can see the device (so I know how to use enroll to pair). That's not my question. After pairing my phone popups a message saying I need a an app to connect.

linglingpin, 2022-07-24:
> Not sure if your comment was towards me, but as I stated: I can see the device (so I know how to use enroll to pair). That's not my question. After pairing my phone popups a message saying I need a [...]

After pairing, you should see the irk in the logs. You can use that to identify your iPhone from now on, and you can also unpair your phone with it.

ASNNetworks, 2022-07-24:
> 

After pairing my phone popups a message saying I need a an app to connect.

PuckStar, 2022-07-24:
> If you're using Home Assistant, the Enroll button is on the ESPresense device page.

Can we also add that button manually somehow?  Because [I couldn't get ESPresense device to automatically show](https://github.com/ESPresense/ESPresense/discussions/564) up in Home Assistant and had to do it manually.

jasiowy, 2022-07-24:
I paird samsung note 10+ and everything is working fine. He sends continuous messages in MQTT topic. Unfortunately, samsung s20 fe after pairing sends messages only when I use bluetooth, e.g. by connecting the speakers. Doesn't send messages continuously.

PuckStar, 2022-07-24:
I'll repeat myself just so my idea doesn't get lost: 
Wouldn't it be possible to just "see" bluetooth devices (on MAC address) just like [Monitor ](https://github.com/andrewjfreyer/monitor) does?

Then we don't need to pair anything and it works well.

ASNNetworks, 2022-07-24:
> I'll repeat myself just so my idea doesn't get lost: Wouldn't it be possible to just "see" bluetooth devices (on MAC address) just like [Monitor ](https://github.com/andrewjfreyer/monitor) does?
> 
> Then we don't need to pair anything and it works well.

That's litterally the whole point of ESPresense, have you even looked at the readme and setup? The issue is that some devices (mainly iPhones and some Android phones) use random mac addresses, which makes other scanning options needed, like pairing or an app.

PuckStar, 2022-07-24:
> That's litterally the whole point of ESPresense, have you even looked at the readme and setup? The issue is that some devices (mainly iPhones and some Android phones) use random mac addresses, which [...]

Yes I did read it and understand how it works. And yes I read that about random addresses. 
But what I was trying to say is that apparently the "Monitor" tool is able to track those kind of devices just fine. So without pairing or an additional application it apparently is "seeing" the fixed Bluetooth MAC address of the device just fine.

As example my Samsung A50 is perfectly tracked by "Monitor" however ESPresense seems to look at something else, namely the randomized ID. Perhaps the randomized ID's is part of the BLE broadcasting while my device already also can broadcast a fixed MAC address.

So that is why I am wondering if ESPresense could also not as well look at that fixed Bluetooth MAC address. I hope you understand what I'm saying.

DanielBaulig, 2022-08-06:
The "heart rate monitor" name should include the set ESPresense room name, to easier identify and distinguish by different paired ESPresense sensors.

DanielBaulig, 2022-08-06:
Note: when I try to enroll my Apple Watch with the ESP connected via serial bus, I get a "Encrypt connection failed conn" message. No `irk:<id>` or `settings/alias` entries show up in mqtt.

DTTerastar, 2022-08-12:
The reason I don't add the room name is BLE has very tight limits on the number of characters and you really aren't pairing to a particular room, we send it to all devices via mqtt.

Apple watches refuse to bind securely.  I tried long and hard... But you can discover your watches IRK from the iCloud Keychain.  #580 has some info on how to do it.

blacknell, 2022-08-14:
The enrol worked like a charm on my iPhone 13. But when I then went to so same on my wife's iPhone X I get nothing in MQTT `espresense/rooms` or `espresense/devices`.

I paired her phone to one of my espresense instances fine (just as with mine). But then nothing in MQTT `rooms` or `devices` showing a new `irk:` entry. I've tried a few times.

I do get a new entry under `espresense/settings` with an empty config (mine is there too, but with a config id equally my irk)

It's rather confusing. Is there some espresense documentation we can review and/or contribute to?

blacknell, 2022-08-14:
also, is the irk entry on the device html config screen a list of known irks against which to filter? ie an allow list?

DTTerastar, 2022-08-18:
yes, the whole thing is confusing atm.  I am working on making the GUI make enough sense that extensive documentation won't be needed.

DTTerastar, 2022-08-18:
> also, is the irk entry on the device html config screen a list of known irks against which to filter? ie an allow list?

No it's just a way of putting in a known irk.  I may delete it because I plan on adding a whole gui to make it possible to add a known irk on the "devices" page.

blacknell, 2022-08-19:
I now have a 3rd device (iphone 6s) added but not showing up on `espresense/devices`. It too published an empty document to `espresense/settings`

Once again the pairing was very straightforward.

I'm not clear on what the `espresense/settings` documents are used for

blacknell, 2022-08-21:
> 

I can't explain why the document is empty but when I put the irk into the document (as I suspect it should be), and reboot the various ESP devices, then they start being shown up in rooms/devices

blacknell, 2022-08-21:
On a related note, is there a way to reboot all ESPpresense devices in the network at once. Do all the devices subscribe to a group topic (such as they can do in tasmota) so I can broadcast a `/reboot` payload

blacknell, 2022-08-24:
.

> 

@DTTerastar should i create a new issue to reflect this bug? I can't add the bug tag to this issue

DTTerastar, 2022-08-24:
> On a related note, is there a way to reboot all ESPpresense devices in the network at once. Do all the devices subscribe to a group topic (such as they can do in tasmota) so I can broadcast a `/reboot` payload

Yes, use room as * and they'll all do it.

closed 2022-09-01 by DTTerastar