ESPresense/ESPresense #574: Iphone Tracking Problem

onuralpp, 2022-08-04:
Hey everyone,
I was using my iphone 11 pro to track inside the home and it was working very good. One day my iphone's bluetooth device id changed automatically. I updated the device id but now its only transmits when the screen is on (it does not have to be unlocked, just on). I couldn't fix the problem.
Does anyone know the solution? 
Thank you

closed 2022-08-05 by onuralpp

nickelnine, 2022-08-05:
What was your solution? I'm facing something similar

BogdanValea, 2022-08-11:
hello guys, same issue on iPhone 12 and iPhone 13, IOS: 15.6 and i don't think is a fix for this one.
ESPresense is working fine.... I'm watching the Terminal and when iPhone screen turn off is not sending BT announcements.

nickelnine, 2022-09-20:
Have you tried setting Auto Lock to Never? It keeps transmitting that way for me but leads to quicker battery drain.

lxgenie, 2023-03-30:
I'm also seeing this same behaviour on iPhone 13 running iOS 16.3.1.

Running ESPresense v3.2.4 and using the IRK as the id which is what is being tracked in Home Assistant, and I've also added the IRK into the known BLE identity resolving keys on all four base stations I'm using.

jjbrunton, 2023-04-02:
Also seeing this on an iPhone 11 and iPhone SE. Both reported as not_home after being locked.