ESPresense/ESPresense #580: Two Apple Watches reporting same AppleID.

Buckeyes1995, 2022-08-08:
Hello,
 
Both my wife's and my Apple Watches report the same Apple ID "apple:1005:9-24".  I'm certain this is happening as I've moved one of them all the way across the house and easily see that both, when by the ESP32 are the correct distance away.  How else can I identify them uniquely?

Thanks,
Jim

Buckeyes1995, 2022-08-10:
Hi DTTerastar,

Doesn't the watch have to be paired to the PC to see it?  There's no way (that I know of) to pair an apple watch to anything other than the iPhone.  

I saw the question you asked about an app to do it over at room-assistant.  I can confirm that we have 2-3 model years apart with our watches and they still broadcast the ID the same.

DTTerastar, 2022-08-12:
Yes, the apple watch only pairs with iPhones.  But it stores in your iCloud Keychain the identity resolving key (irk) that is used.  You can take that from the keychain and add it to ESPresense.  My watch is setup and works great and it's 100% unique.

Buckeyes1995, 2022-08-12:
It's weird.  Bluetooth is on, the KEYS folder is completely empty.  I ran PsExec -i RegEdit as suggested and it's still blank.   Windows 11.

Buckeyes1995, 2022-08-15:
Glad you got it working, I am not able to see any keys in my keychain via Regedit.  

Anyways, you need to add it to your sensor, such as:

`- platform: mqtt_room
  device_id: "irk:7ea0791432blahblahblah"
  name: 'Beacon - Jims Iphone'
  state_topic: 'espresense/rooms'
  timeout: 30
  away_timeout: 120`

ikolsteren, 2022-08-15:
I was missing the irk and had dashes in the ID but looks like it isnt the code i need.
So in the same boat about how to get the Apple watch IRK code
But I am also trying to get it from a mac only - no windows

Buckeyes1995, 2022-08-16:
Yeah, hoping DTTerastar updates the docs to be a little more clear.  I know he's done a ton of work and it's pretty obvious to him, but it can be a bit daunting.

As far as doing it only on Mac, not sure.  Check the thread that he linked in reply to my opening this issue.

I am getting very inconsistent behavior with my iphone.  It runs a while and then I literally have to switch bluetooth off and back on to get it to start updating again.

DTTerastar, 2022-08-18:
Sorry, yeah docs are only in github issues atm.  The irk you discover is meant to be put into the "Known irks" in the main config.  A further wrinkle is the IRKs have endianess and there isn't a clear standard.  I want to just put both variations in when you put in a known irk in order to work around that.

Buckeyes1995, 2022-08-28:
Was able to resolve following the instructions here: [https://github.com/ESPresense/ESPresense.github.io/pull/33](url)

closed 2022-09-01 by DTTerastar