ESPresense/ESPresense #586: WiFi Disconnects from initial AP

charredchar, 2022-08-13:
I am using a ESP32-DevKitC board with version 2.2.9, but I also attempted 2.2.8 and 2.2.7 with similar results. After installing the firmware and the device rebooting, showing an AP is created in the terminal, when ever I use my cell phone (the only WiFi device I have available) to load the page for setup it'll disconnect from the WiFi connection. The log shows it finds WiFi networks to connect to but nothing else in the log. When I reconnect with my phone the same thing happens over and over again. I got lucky with one device and it stayed loaded just long enough for me to get my WiFi password in and that device has been running fine since but I can't get any more set up.

DTTerastar, 2022-08-13:
While connected go to http://captive.apple.com in a web browser.  Can you see set it up that way?

charredchar, 2022-08-15:
Yes and no. It forwards to the setup page as you need to "Sign in to WiFi Network" but it will continue to disconnect and reconnect in the background. The only difference here is when it disconnects it doesn't force close the browser window. If I can get the entire page to load before a disconnect I can then enter all my settings and while connected again I can save and restart. I confirmed it does save and the settings take effect, once it is on my home WiFi I have zero issues with the pages. It also doesn't have to be the Apple URL, any works, even the default 192.168.4.1.

I also tried this with v3.0.0b6 with the same results. I also tried a different set of ESP32 devices, the ESP-32S.

armdan, 2022-08-19:
Exactly same thing is happening to me . I also tried installing  v3.0.0B6 with no luck. I installed Tasmota, EspHome and it worked fine but not ESPresense

DTTerastar, 2022-08-23:
Can you get a log of serial output while this is happening? (https://espresense.com/terminal)

DTTerastar, 2022-09-01:
We now kinda support the improv serial.  It get's stuck but it works for me, can you try that w/ v3?

closed 2023-02-14 by stale[bot]