ESPresense/ESPresense #589: Is it possible to consume ESPresense as an Arduino Library?

TonyBrobston, 2022-08-13:
**Is your feature request related to a problem? Please describe.**
I am working on an MQTT Smart Vent project that I plan to make public soon (finishing up documentation and determining licensing). The over-arching system brings together a thermostat, temperature sensor, and smart vents to make your home more comfortable. Naturally my mind gravitates to only heating/cooling occupied rooms. From what I can tell, I could buy an ESP32, load ESPresense, and work the "presense" concept into the over-arching system as a feature; however the downside is that running another ESP device adds an extra component to the system. Is ESPresense consumable as an Arduino Library? If not, is this theoretically possible? 

**Describe the solution you'd like**
I would like ESPresence to be consumable as an Arduino Library so that I can put an ESP32 into my Smart Vent project and kill two birds with one ESP.

**Describe alternatives you've considered**
Run ESPresence on an ESP32 and try to figure out how to justify another component in the system.

**Additional context**
Thanks to all the contributors for the work they've done on this project, it's an awesome concept!

DTTerastar, 2022-08-13:
Super cool!  I'd suggest going the other way and just adding your vent to this codebase.  You could fork it and just add a Vent.cpp / Vent.h and put all the logic in there.

DTTerastar, 2022-09-01:
I am totally going to print one of these one of these days...

closed 2022-11-08 by stale[bot]