ESPresense/ESPresense #593: Support for Hidden WiFi

idaence-dfi, 2022-08-14:
Hi, the current Version of ESPresence only supports named Wifis. I have different hidden Wifis in my Home and I want to see an Option to insert my Wifi Configuration manual. 

there was a Pull Request on ESP-Wifisettings but I think the WiFi engine has changed on the Version 3.
https://github.com/ESPresense/ESP-WiFiSettings/pull/1

Is their any option to add hidden wifis to the Configuration Page? 

I hope I would love to see this option on a future release, if there is any other way i can insert my wifi configuration i would test them in my network.

thanks and regards,
Daniel

closed 2022-08-23 by DTTerastar

DTTerastar, 2022-08-23:
The latest beta allows you to select Wifi SSID manually after flashing via webtools.

idaence-dfi, 2022-08-27:
Thanks, works like a charm! :) Tested on 6 Devices! :)

jonny190, 2024-06-18:
> The latest beta allows you to select Wifi SSID manually after flashing via webtools.

Ive just tried this but on the ESP32 C3 when the device connects to windows it shows as a unknown device 
[image]

emilianox, 2025-01-03:
Hi @idaence-dfi 

I saw your comment form connecting to a hidden SSDI. I’m trying to do the same but can’t find this option.

is there any chance that you please provide more details how to do it?

Thanks in advance!

jonny190, 2025-01-03:
I managed via Matt commands ... However I ended up switching to esphome and Bermuda plugin in home assistant

emilianox, 2025-01-04:
@jonny190  thanks man, really appreciated!

commit 07228feb (2022-08-23): Selectable LEDs and Update Button (#595)

* RGB LEDs supported (closes #297) 
* Add button to manually update (closes #533)
* Manual Wifi setup via web installer (closes #593, #172)
* Add esp32c3-verbose flavor (#134)
* Make scan all wifi the default (#594)
* Improvements to M5Atom/M5StickCs