ESPresense/ESPresense #594: Wifi fails to connect after upgrade from 2.1.x

TechIsCool, 2022-08-14:
Figured I would get back to upgrading my esp-presence devices to the latest version to see if it was performing better. When flashing the firmware it succeeds but then I can no longer connect to my wifi.

I have tried a bunch of different things before filing a bug repot
  - Upgrade In Place without WIPE
  - Fully Wipe NVS
  - Verbose Latest GA
  - Verbose Latest Beta
  - Create a New Wifi Network with no space in it
    - Change to WPA AES
    - Change to WPA TKIP
  - Restore back to 2.1.x works

Each of these devices can be flashed with ESPHome and work on the network. 

Details:

**Router**
- MikroTik hAP
- Version: 7.2.3
- SSID "TechIsCool Devices"
- WPA, WPA2, AES, TKIP
- Band: 2GHZ B/G/N
- Width 20/40 Mhz Ce
- Country US 2.4 5.8

**Config**

- Version: 2.1.x+
- Active scan enabled: true/false
- Include filter: n/a
- Exclude filter: n/a
- Query filter: n/a

**Hardware**

- Firmware flavor:
  - Verbose Latest GA
  - Verbose Latest Beta
- Device (be specific):

ESP32-ROVER-E
ESP32-WROOM

**Logs**

2.1.1 Verbose
```
Setup network
I (80) wifi:wifi driver task: 3ffd5a18, prio:23, stack:3584, core=0
I (392) wifi:wifi firmware version: dc30037
I (393) wifi:config NVS flash: enabled
I (393) wifi:config nano formating: disabled
I (394) wifi:Init data frame dynamic rx buffer num: 32
I (398) wifi:Init management frame dynamic rx buffer num: 32
I (404) wifi:Init management short buffer num: 32
I (408) wifi:Init dynamic tx buffer num: 32
I (412) wifi:Init static rx buffer size: 1600
I (416) wifi:Init static rx buffer num: 16
I (420) wifi:Init dynamic rx buffer num: 32
I (514) wifi:mode : sta (24:0a:c4:5b:98:e0)
Connecting to WiFi SSID 'TechIsCool Devices'..............I (2581) wifi:new:<4,1>, old:<1,0>, ap:<255,255>, sta:<4,1>, prof:1
I (2583) wifi:state: init -> auth (b0)
I (2588) wifi:state: auth -> assoc (0)
I (2595) wifi:state: assoc -> run (10)
I (2611) wifi:connected with TechIsCool Devices, aid = 16, channel 4, 40U, bssid = a1:8e:f3:ba:2f:4a
I (2612) wifi:security type: 2, phy: bgn, rssi: -41
I (2615) wifi:pm start, type: 1

.I (2688) wifi:AP's beacon interval = 102400 us, DTIM period = 1
192.168.1.88
Firmware:     esp32-verbose
Version:      v2.1.1
IP address:   192.168.1.88
DNS address:  192.168.1.1
Hostname:     espresense-livingroom
Room:         livingroom
MQTT server:  192.168.1.10:1883
Max Distance: 3.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
```

2.1.2 Verbose
```
Setup network
Connecting to WiFi SSID 'TechIsCool Devices'.................192.168.1.88
Firmware:     esp32
Version:      v2.1.2
IP address:   192.168.1.88
DNS address:  192.168.1.1
Hostname:     espresense-livingroom
Room:         livingroom
MQTT server:  192.168.1.10:1883
Max Distance: 3.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:  
```

GA Current Release
```
Connecting to WiFi SSID 'TechIsCool Devices'..............[  2600][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[  4660][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[  6720][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.............[  8781][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 10841][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 12900][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 14960][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.............[ 17019][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 19078][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 21137][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.............[ 23196][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 25257][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 27316][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 29376][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.............[ 31435][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 33494][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 35553][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............[ 37613][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.............[ 39672][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
...... failed.
Starting access point for configuration portal.
```
**Screenshots**

If applicable, add screenshots to help explain your problem.

DTTerastar, 2022-09-01:
Are you still broke?  I did mess w/ a few WiFI settings during the beta, I think we're all good now, but let me know.

TechIsCool, 2022-09-04:
3.0.4 Verbose, I have tried both with connecting via the portal and also configuring using the serial UI for setting WIFI.

```rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID 'TechIsCool Devices'..........................................[  2664][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[  4720][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[  6776][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[  8830][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 10886][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 12942][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 14998][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 17053][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 19108][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 21162][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
..........................................[ 23217][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 25272][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 27326][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 29381][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 31437][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 33492][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 35548][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 37603][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.........................................[ 39657][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
................... failed.
Starting access point for configuration portal.
SSID: 'espresense-08b9bd'
IP: 192.168.4.1
10 WiFi networks found.
```

3.0.0 Beta 11 also give the exact same results. I even tried upgrading my Wireless firmware to see if there is a bug but all other ESPHome devices stay connected.

DTTerastar, 2022-09-05:
Another user found #614 does that fit your issue?

TechIsCool, 2022-09-05:
Password doesn't contain trailing `!` in it, also its worth mentioning that I also tried with an unsecure SSID, `esp-presense` and it still didn't connect.

DTTerastar, 2022-09-05:
What WiFi channel?  #628

TechIsCool, 2022-09-05:
```2442/20-Ce/gn``` AKA channel 7, I have pushed my Wifi over to channel to ```2412/20-Ce/gn``` AKA channel 1 and we will see if this changes. 

Nope no change, currently the https://espresense.com website is down. So I can't flash the build in the [linked branch](https://github.com/ESPresense/ESPresense/actions/runs/2996342317) correctly. I did try using https://web.esphome.io but that didn't work either.

DTTerastar, 2022-09-05:
https://github.com/espressif/arduino-esp32/issues/6654 this seems to track w/ your experiences, might want to pipe up in there.  Arduino 2 is new in [v3.0.0b3](https://github.com/ESPresense/ESPresense/releases/tag/v3.0.0b3)

https://espresense.com/firmware is up for me

DTTerastar, 2022-09-06:
I have now experienced the website going up and down.  I needed to re-up the domain but yet it continues.  Is it up for you now?

closed 2022-11-13 by stale[bot]