ESPresense/ESPresense #596: ESP32-WROOM stuck in reboot loop

hanfreakingsolo, 2022-08-16:
**Describe the bug**

I have 3 ESP32-WROOM devices running espresense. The one in my kitchen is stuck in a constant reboot loop, and it seems to be related to memory usage. I've tried replacing it 3 times with other ESP32s in case it was a hardware issue, but same problem. In home assistant, the free memory drops steadily for a few minutes until the device decides to reboot itself. Memory jumps back up, then falls again. And over and over it goes. The device in my bedroom doesn't do this - the free memory graph is pretty stable, goes up and down a bit but no major spikes. The one in my living room is somewhere in between; reboots occasionally when free memory drops.

**Config**

- Version: 2.2.9
- Active scan enabled: No
- Include filter: Usually none, tried limiting to only include one device and problem persists
- Exclude filter: N/A
- Query filter: N/A

**Hardware**

- Firmware flavor: Standard
- Device (be specific): ESP-32S (ESP-WROOM-32)

**Logs**

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID '******'*******192.168.108.102
Firmware:     esp32
Version:      v2.2.9
IP address:   192.168.108.102
DNS address:  192.168.108.1
Hostname:     espresense-kitchen
Room:         Kitchen
MQTT server:  192.168.1.5:1883
Max Distance: 5.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
1 New   | MAC: d44da42b0431, ID: d44da42b0431                                              -88dBm 
1 New   | MAC: 63b3bceca396, ID: msft:cdp:0906                                             -88dBm 96f7d933d376a46e5fc1c2fd8ee47017af89081f

*** Removed a ton of devices here to make reading logs easier; well over a hundred newly discovered beacons ***

1 New   | MAC: 65ca86d362cd, ID: apple:1007:11-12                                          -93dBm 1a1fa3ec3b9268
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID '****'****.192.168.108.102
Firmware:     esp32
Version:      v2.2.9
IP address:   192.168.108.102
DNS address:  192.168.108.1
Hostname:     espresense-kitchen
Room:         Kitchen
MQTT server:  192.168.1.5:1883
Max Distance: 5.00
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
1 New   | MAC: 009d6b61520b, ID: 009d6b61520b                                              -76dBm 
1 New   | MAC: 4293cf532519, ID: iBeacon:587eb13e-4ad3-4968-ab84-2e29c888b64d-100-1        -56dBm 
1 New   | MAC: 47214ffbd8b4, ID: apple:1007:11-7                                           -72dBm 2a1f6c5610f928
1 New   | MAC: 5cc1d79950bf, ID: 5cc1d79950bf                                              -95dBm 
1 New   | MAC: 63b3bceca396, ID: msft:cdp:0906                                             -92dBm 96f7d933d376a46e5fc1c2fd8ee47017af89081f
1 New   | MAC: 51f0af3bd1d2, ID: msft:cdp:0902                                             -43dBm b4601d3821d9d00f09882e3596a93f43ad743290

**Screenshots**

[image]

DTTerastar, 2022-08-30:
v3 is out and does a bunch to attempt to make memory usage better.  If you're running low on memory it won't check for updates (you can still manually check for updates and it reboots before applying updates so lots of ble traffic won't break it)  Can you test it and let me know?

hanfreakingsolo, 2022-09-01:
Upgraded to v3, and it's definitely better... It lasted a few days without crashing from memory usage until today. My kitchen sensor has now gone into a reboot loop with the same memory spikes

DTTerastar, 2022-09-01:
Have you turned off auto updating?  It uses like 70k of memory.  WROVERs have lot's more memory, i only have 2 but they are super rock solid because of the extra memory.

hanfreakingsolo, 2022-09-02:
Turned off auto updating an so far not seeing much difference, although I'll need to leave it running for some time to see. My apartment is right next to a very busy street, and I suspect large numbers of people walking by BLE devices are what's causing issues as the crashes are usually during the busiest parts of the day. I avoided this temporarily by reducing the 'forget beacon if not seen' time so it's not hanging on to these devices, but had to reduce it too low and it was forgetting my own devices. Will follow up

DTTerastar, 2022-09-05:
Have you tried using  ROVER module?  They're a few bucks more, but the extra memory really helps.

DTTerastar, 2022-09-06:
https://www.aliexpress.com/item/3256801924539744.html?spm=a2g0o.order_list.0.0.3ac31802lqUEZZ These are also great for memory, WiFi uses a bunch of memory and using the Ethernet frees up a ton.

hanfreakingsolo, 2022-09-07:
Thanks for the link, maybe I'll have to look into that. Playing with forget beacon settings to try and find a balance before running wires.
I only have one ROVER module available I was hoping to use for another project, I ordered 8 WROOM modules specifically for espresence.. wish I knew about the memory use in advance. Maybe a suggestion to add a disclaimer to the supported devices list suggesting more memory for those in high-traffic areas?
So far the crashes are infrequent enough I think it's not a huge issue anymore (once a day roughly), v3 definitely made an improvement! If memory use continues to improve then I think the WROOM will be just fine.

closed 2022-11-14 by stale[bot]