ESPresense/ESPresense #601: Automatic downgrade?

wenjiavv, 2022-08-21:
I upgrade the  EPS32 to the lastest pre-release version (3.0.0b6). The configuration page shows the version is 3.0.0b6 and i fiil the blank in Sacn section with my iphone IRK. it works perfectly after restart,.The mqtt is connected and HA shows the correct loaciotn.

After all set, i deplug the ESP32 and put it somewhere else in my livingroom and plug again, then i leave the room. When i return the room i find the status of iphone keep showing not_home. i check the configuration page and find the **version is v2.2.9**, and the IRK blank is missing. However in MQTT, it still shows v3.0.0b6.

i am not true if it's a bug or i just did something wrong. 

[image]

[image]

DTTerastar, 2022-08-23:
Turn off auto update OR turn on allow prerelease.  That should fix it.

wenjiavv, 2022-08-23:
> 

thanks a lot. it works perfectly now.

closed 2022-08-23 by DTTerastar