ESPresense/ESPresense #602: Documentation lacks on iOS devices

shakisha, 2022-08-22:
I'm frustrated when it comes to iOS devices and domoticz platform, there is no documentation about.
I need to understand how to find my iOS device after the firmware flashing, I cannot understand if I need to install an app or if I can use.....

I would like an help to write a tutorial, all the quick start videos are only about android devices.

DTTerastar, 2022-08-23:
Once the new beta drops there will be one way that works well.  At the moment there are in fact 3 sorta ok ways.  Will get back to you.

shakisha, 2022-08-23:
@DTTerastar yes, I have tried the irk way with pairing, but it is sometimes failing without any reason, so my mobile is constantly seen and not seen.

antongiacomo, 2022-08-25:
[Here](https://github.com/ESPresense/ESPresense/issues/460#issuecomment-1220287957) there is a fast step by step. I noticed in that message there is a "reverse" step which is missing here. You could try!

fila612, 2022-08-25:
Thanks, but what is meinst vielen Dank reverse ist?
Of the irl is „abcde“, should it used as „edcba“?
Or should is reverse en-/decoding step needed?
I don‘t understand the reverse step….

antongiacomo, 2022-08-25:
> Thanks, but what is meant with reverse step? If the irk is „abcde“, should it used as „edcba“? Or is reverse a additional en-/decoding step needed? I don‘t understand the reverse step….

It should be what you said, if the string is "abcde" then it becomes "edcba".

The new beta version allow also to perform the enroll procedure. If you can follow this path you can avoid all the keychain thing, and get directly the IRK .

fila612, 2022-08-25:
okay, I've tried to use the reversed irk, but unfortunately its not working.
with the "apple:1000?..." code, I used to point on the state topic 'espresense/devices/apple:100???..".
with the irk its, not working, because in the device topic are only the "apple:100??...." things.
so i chanced the state topic to 'espresense/rooms', but it's also not working: no information in the MQTT Tool and no activity in HA.

I used the beta 3.0.0.b8, is there you're mentioned option available?

antongiacomo, 2022-08-25:
Yes it should be there, i followed [these](https://github.com/ESPresense/ESPresense/issues/460#issuecomment-1190682257) steps and i got the IRK.  In the comments below you can find also an example of configuration. 

I've been able to setup the thing, and know i get some sort of sensible result. I'm still testing because the device tends to disappear.

fila612, 2022-08-25:
hmm, okay - thank you.
the point is, that I've had also tried theses steps and I got no irk (don't no why?).
so I got my irk via KeyChain (hopefully is the right one, maybe this could be the error).
geh examples are known and used in my HA, but nor working.
i will test later, thanks for your help.

antongiacomo, 2022-08-25:
Can you share your configuration?  Mine should be correct, but the HA still shows "not_home"

fila612, 2022-08-25:
- platform: mqtt_room
  device_id: "irk:fe1426e26f51c29cf62e65b46371b031" 
  name: 'iPad Pro'
  state_topic: 'espresense/rooms'
  timeout: 10
  away_timeout: 120

fila612, 2022-08-26:
so status from myside:
iPhones, iPads seems to be working with the IRK (getting them via enrols-procedure was easy).
my Apple Watch is not able to share the irk, I cannot use the same way like the iPhones. when I connect to espressos after activation enroll, I see on my watch the the connection to espresens is active and connected, but there is no irk.

[06:52:29]1 MQTT  | Set enroll: PRESS
[06:52:29]Advertising started
[06:52:30]Client connected
[06:52:30]Multi-connect support: start advertising
[06:52:30]Client address: f4:d4:88:65:xx:xx
[06:52:30]Client disconnected - start advertising
[06:52:35]Client connected
[06:52:35]Multi-connect support: start advertising
[06:52:35]Client address: 58:97:ef:71:xx:xx
[06:52:36]MTU updated: 23 for connection ID: 0
[06:52:36]Encrypt connection failed conn: 0!
[06:52:36]Client ID: 0 Address: 58:97:ef:71:xx:xx Subscribed to notfications for 0x2a37

58:97:ef:71:xx:xx is the public Mac address of my apple watch.
for now, I think there is now way to use ESPresense to track apple watches.
maybe someone can help or share some hints/experience what's going wrong.

Buckeyes1995, 2022-08-26:
Fila, would you mind explaining a bit more on how you found your irks in KeyChain Access?  What did you search for, an application-password?  If I open up any of the password's I don't see any XML structure.. I do see a 'show password' option.. obviously I'm not sure what I'm looking for.

Thanks.

fila612, 2022-08-26:
okay, as written above, I checked the content of my keychain. so the devices on my machine are stored as a Programm password and the names are a little bit cryptic, like "ECF39ABA-F1D4-92F8-BB4E-3C6F595F57EC".
I had round about 10 of these entries in my keychain.
when you open them, you'll see this window:
[image]
I compared the public MAC with the MAC what I see in the info menu on my watch, they are matching.

so at the bottom, there is a checkbox and if you check them, the textfield will show you some hidden content (after legitimation via password). 
click in this field, and copy the whole content (it's more than the one line you see) I used cmd+A and then cmd-C (or STRG+A & STRG-C) to catch the content.

paste this in a Texteditor and you'll see the xml structure.

Buckeyes1995, 2022-08-27:
Thanks!  I found my watch in the Key Chain, but the IRK doesn't make sense.  Tried to 'Base 64 Decode it" but the online tools just produce garbage.  What's in my file is a 'Remote IRK' key, with data similar to:

MtP+wigLPlXs1x693sRDAQ==

How do I convert that to a 32 bit HEX string?

fila612, 2022-08-27:
Thats the question, don‘t know

antongiacomo, 2022-08-27:
You have to convert it to hexadecimal non to text and then reverse the byte obtained. 
For example my IRK (obtained with the enroll procedures) is something like:   93df......6d.

[image]

Now, as you can see in the screenshot (I'm using [cryptii](https://cryptii.com/pipes/base64-to-hex)).
Taking the base64 from XML, I got the byte ordered from last to first, I just need reverse them byte to byte et voilà.

RaoulSargent, 2022-08-27:
The instructions here work perfectly for Apple Watch. (Assuming you have a Mac).

https://github.com/ESPresense/ESPresense.github.io/pull/33/commits/632175693017a391e80aa7a945c1f0c58558806b 

Take note:
Having found the Base64 string in the Apple Keychain you must first Decode that using the site linked in the above instructions, not the usual Base64decode. 

Secondly you must reverse the order of PAIRS of characters, not every character in reverse. 
So A1B2C3 becomes C3B2A1

I followed those instructions and everything works with HA and my Apple Watch (x3 in family)

I think the mistakes that are being made are not correctly decoding the base64 and not correctly reversing the bytes.

fila612, 2022-08-27:
Guys-your are awesome!!!!
Now it‘s working-thank you so much.

my mistake was to reverse Not the PAIRS and i was using the system entry of my Watch- Not the icloud entry

antongiacomo, 2022-08-27:
@fila612  Is your apple devices detection stable? I'have a lot of "not_home" reports

fila612, 2022-08-28:
hm, it depends :)
my Apple Watch is since I'm using the IRK more stable than by using the normal ID.
but, my iPhone is more "not_home" as the watch.
So, I cannot explain why, but It can be the reason that I have to tweak the settings regarding absorption and distance.
e.g. I saw in the MQTT-explorer, that devices were in 2 rooms and sometimes (by moving around) the devices are in no room (=> not_home).

Buckeyes1995, 2022-08-28:
Wow, thanks for the detailed instructions! Hopefully they find their way into a merge soon!  Apple Watch detected!

Too early, but updating to b10 and now my phone is consistently reporting... weird.

Buckeyes1995, 2022-08-28:
Another weird issue.  I added my wife's Apple Watch.  I can see it sending MQTT messages but Home Assistant always shows it as not_home.  I can see the distance is within the filter range.  Very confused here.  Any ideas?

janssenjones, 2022-08-30:
> @fila612 Is your apple devices detection stable? I'have a lot of "not_home" reports

I've been finding the page at http://192.168.x.x/devices#fingerprints to be very helpful. It's basically a webpage version of what's getting sent over to MQTT.  It's not very easy/useful to troubleshot at the Home Assistant layer.  You sort of have to go upstream to either the devices or to MQTT.  

I used the new pairing feature to pair all of my devices and get their IRKs, and now what I've been seeing, is that some of the time, my phones are getting reported to MQTT by their IRK and some of the time they are getting reported to MQTT by their friendly device name.  I've now got a console that shows all six rooms and what IDs they are seeing, and that's helping to narrow things down and see where the anomalies are.

DTTerastar, 2022-08-30:
I moved /devices/ to /ui/ in the release.  Docs are updated, can we close this?

janssenjones, 2022-08-30:
> I moved /devices/ to /ui/ in the release. Docs are updated, can we close this?

I'm confused here.  I'm on 3.0.1, but I only have /devices and /devices#/fingerprints
/ui returns a 404.

closed 2022-11-05 by stale[bot]