ESPresense/ESPresense #603: Tile sticker problem

rbjoensen, 2022-08-22:
Hi, I've been using a Tile sticker on my dog's collar for the past few months, it's been stable.
But the last few weeks it has started to be very unstable and not repoting correctly. If I place my Tile 1 meter from the esp unit it can repot not_home
I then thought my Tile was not working properly so I bought a new one, but it does the same thing.
Is there anything I can do to eliminate the problem.

This is the Tile I use https://uk.tile.com/product/687312/sticker-1-pack

kind regards RBJ

webhostsg, 2022-08-22:
@rbjoensen I've noticed that same thing in the past few weeks after it auto upgrades to Firmware: v2.2.9. I need to reboot the ESP to get it to detect my tags. Maybe you could try that to see if it's the same issue. I also noticed that when I unplug the ESP for some time the radar and motion MQTT messages are retained and I need to remove the retained message using MQTT Explorer manually.

jasiowy, 2022-08-23:
I had the same problem and turning off active scan helped me.

webhostsg, 2022-08-23:
turning off active scan did not help for me

denko81, 2022-08-25:
I just bought Tile sticker and it shows up in the webterminal but it doesn't create a sensor in Home Assistant.

webhostsg, 2022-08-25:
@denko81 have you added them in `configuration.yaml`? 
https://espresense.com/home_assistant

denko81, 2022-08-25:
Yes, I have.

DTTerastar, 2022-08-25:
> @rbjoensen I've noticed that same thing in the past few weeks after it auto upgrades to Firmware: v2.2.9. I need to reboot the ESP to get it to detect my tags. Maybe you could try that to see if [...]

I would try an older firmware until v3 comes out.

webhostsg, 2022-08-26:
> I would try an older firmware until v3 comes out.

I'm using M5Stamp Pico, I need to remove it, unsolder some wires and wire up USB to UART to downgrade. I think I'll just wait for v3 to auto-update.

webhostsg, 2022-08-30:
> I would try an older firmware until v3 comes out.

@DTTerastar updated to v3 still same issue for me

webhostsg, 2022-08-30:
> @DTTerastar updated to v3 still same issue for me

@DTTerastar I think v3.0.1 is fixed it. Thank you

denko81, 2022-08-30:
@DTTerastar is the issue that I have described also solwed now?

denko81, 2022-08-30:
@DTTerastar Do you have any idea what I should do?

DTTerastar, 2022-08-30:
Search the entities page.  It's in there.

denko81, 2022-08-30:
@DTTerastar I found it but wonder how often is tile updating?

DTTerastar, 2022-09-01:
How is v3 working for you?

cromelex, 2022-09-07:
I'm on 3.0.4. and I've **no** issues with the Tile stickers (I have on tag on each of my cats),

I do use filtering so that only a limited number of devices get tracked (and avoiding noise on the MQTT server), not sure if this makes a difference.

If it helps, I see the distance updating for the tracker entity every ~5 seconds for the Tile sticker.

Trevsweb, 2022-09-07:
Having no issues on v3 running 3 tile stickers on cat collars. Running about 8 atoms around the house. I do have home assistant automation rebooting them every night at midnight to maintain performance. But pretty solid now it's setup. Just gotta tune the jitter of picking up wrong overlapping rooms but it's really cool

closed 2022-09-08 by DTTerastar

keplenk, 2023-07-15:
> I'm on 3.0.4. and I've **no** issues with the Tile stickers (I have on tag on each of my cats),
>
> I do use filtering so that only a limited number of devices get tracked (and avoiding noise on the [...]

Hello!  I'm very new to ESPresense and I'll be using Tile Stickers to track my new adopted cats (2 of them).  My whole ESPresense are just for cats so I really only want to filter the two tile colars.

If you are cool with it, can you please send a screenshot of your config on how will the filtering look like?

I was playing with it and can't figure it out for some reason - it still sees a bunch of BT devices.

cromelex, 2023-07-15:
> Hello! I'm very new to ESPresense and I'll be using Tile Stickers to track my new adopted cats (2 of them). My whole ESPresense are just for cats so I really only want to filter the two tile [...]

It's really straight forward. First, without filtering, get the id's for your tiles.

Then, just add them in the filtering field (space separated). I have two tile tags and 1 tile mate which is why it shows 3 different tile IDs.

[image]

Espresense is so reliable that I even managed to make an automation to let me know when the cat is outside trying to get back into the house - she has ran away a couple of times and rather than keeping a door open, once espresense detects the tile, I'm getting a notification via HASS to go and open the door to let her back in.
Let me know if you need any more information.

keplenk, 2023-07-16:
> It's really straight forward. First, without filtering, get the id's for your tiles.
>
> Then, just add them in the filtering field (space separated). I have two tile tags and 1 tile mate which is why [...]

Thank you for this, I just tried it and it works! :)  I was doing it wrong.