ESPresense/ESPresense #609: PWM seems to be always PWM inverted in v3.0.0b9

antongiacomo, 2022-08-25:
I have a led connected on the gpo13. The control of the led by home assistant works but is reversed. if I set the brightness to 0 I get it maximum and vice versa. The same happens by turning the LED off and on, if I turn it off it turns on

Changing the setting from PWM to PWM Inverted does not seem to have any effect

**Hardware**

- Firmware flavor: Standard
- Device (be specific): ESP32-Cam

**Logs**

I cannot attach right now, but I'll update the issue if needed

**Screenshots**

[image]

DTTerastar, 2022-08-30:
Yup, can confirm

antongiacomo, 2022-08-30:
I've tried to spot the problem but what I saw seemed okay.

DTTerastar, 2022-08-30:
I figured it out, it'll be in the next release.

closed 2022-08-30 by DTTerastar