ESPresense/ESPresense #614: Wifi connection failed when special character is used

Diddlik, 2022-08-25:
I did several tests and found,

if I have a "!" character at the end of my WiFi password, then esp32 does not connect to WiFi. It is important that ! is at the end, if it is in the middle of the password, then everything works.
Does not work: abcdeF45!
Goes: abcd!eF45

DTTerastar, 2022-09-01:
Man I have ! at the end of almost all my passwords.  That's got to work!

DTTerastar, 2022-09-01:
I just made me a IOT wlan with a password that ends in ! and guess what, it worked perfectly!  I'm not sure what I can unless others experience it.

janssenjones, 2022-09-02:
I just wanted to confirm that I have six of these running, and every single one connects to an SSID that has a password ending with an exclamation point.  There must be something else happening...

DTTerastar, 2022-09-02:
Passwords with ! at the end are the best :)

Diddlik, 2022-09-19:
I have an update to this bug. If I setup espresence WiFi with Web Tool (after Flashing, they ask me to setup WIFI), in this case WiFi with special character works fine. If I setup WiFi over Access Point from ESP32 (after flashing) than it does not work.

closed 2022-11-25 by stale[bot]