ESPresense/ESPresense #616: how to force update from b7 to b10

blacknell, 2022-08-26:
The code on the device is v3.0.0b7 so the automatic updates to v3.0.0b10 aren't going to work (as b7 was broken). So I've remove pre-release and then intended to switch that back on again. But the release binary isn't installing either.

Relates #610

onlyoneme, 2022-08-26:
Similar issue with b8. No automatic upgrade nor downgrade is working.

DTTerastar, 2022-08-29:
Yeah, based on breaking updating so many times I've made a new way to run/auto update PRs so this hopefully can be avoided in the future.  But for now a serial cable is required for updating these beta versions.

DTTerastar, 2022-08-30:
I was able to get all my b7/b8s to respond to espota.py using the method outlined in this issue: https://github.com/ESPresense/ESPresense/issues/619

closed 2022-08-30 by DTTerastar